turing-machines/BMC-Firmware #1: Initial Rust support

ruslashev, 2023-04-10:
To start writing components in Rust, we first need to prepare a library to be built by buildroot and linked in by the firmware.

CFSworks, 2023-04-19:
What is the justification for making it a (C-callable, apparently) library as opposed to other approaches? (e.g. daemon that exposes an API, or a collection of services on an embedded messaging bus, ...)

Many of the functions need to be managed exclusively by a single process, so trying to share this code across many processes seems dangerous.

svenrademakers, 2023-04-20:
You see here the very beginnings of a new feature-packed BMC firmware.
We are in the very early stages, so bear with us :)

The current goal is to gradually switch over from the old architecture to the new one which we are currently drafting. This means that over the course of the upcoming months, we will support some use cases that we will break down soon after.

therefore consider this fork "very very unstable"

CFSworks, 2023-04-20:
Hi Sven! Thank you for your reply.

> You see here the very beginnings of a new feature-packed BMC firmware.
> We are in the very early stages, so bear with us :)

'tis why I'm here now. It's exciting stuff!

By the way, I'm willing to do PRs for pretty much any of the (very many) other issues I've opened here, but I figure they should be discussed before I invest the effort. I don't want to do something contrary to the overall plan.

> The current goal is to gradually switch over from the old architecture to the new one which we are currently drafting.

I'm in agreement that a piecemeal, component-by-component migration does sound like the wise thing to do. Though, my question remains: why a Rust library to (apparently) support the C daemon, rather than replacing the whole C daemon with a Rust equivalent? Shouldn't the `bmc` binary be the first component to be upgraded?

What's the current plan here?

svenrademakers, 2023-04-23:
Thanks for taking an interest in our work. I don't want to go into too much design detail, as this is not the correct place for that.

First objective is to extend the current api with functionality to flash nodes, as this is missing.
New functionality will therefore be compiled into the BMC app.

I will triage your issues shortly, issues regarding the current API i will most likely decline. Efforts to improve an obsolete API is not an area i would want to spent time.

H1ghBre4k3r, 2023-05-20:
I would be more than ready to help implementing some features in Rust/help with the bootstrapping process. 

A small POC written in Rust for powering nodes on and off already works. Furthermore, I think, that the type and memory safety of Rust together with the zero cost abstractions of the language would fit perfectly for a BMC firmware.

svenrademakers, 2023-05-22:
Nice to meet you @H1ghBre4k3r!
We are progressing nicely at the moment and will soon open a PR with an implementation based on gpiod. Maybe you could take a look at it as soon as it opens?

svenrademakers, 2023-05-23:
Our work supersedes this ticket as part of #22. Therefore i will close it. tracking of related work will be done as part of of #22 .

commit 882a5f17 (2023-04-10): app, buildroot: add initial version of tpi_common

Fixes #1
commit 5c4fb810 (2023-04-12): app, buildroot: add initial version of tpi_common

Fixes #1
commit a7eae4f3 (2023-04-21): app, buildroot: add initial version of tpi_common

Fixes #1
commit c0a01ee1 (2023-05-06): app, buildroot: add initial version of tpi_common

Fixes #1
closed 2023-05-23 by svenrademakers