turing-machines/BMC-Firmware #10: Let Linux manage ATX power and node VRMs through its own power management system?

CFSworks, 2023-04-19:
The Linux power management subsystem has the ability to manage power regulators connected to GPIO pins. By creating `regulator-fixed` nodes in the device tree, we can have Linux itself manage ATX and VRM power state. Linux's PM code is also dependency-aware and can understand that the ATX power is a prerequisite for bringing up node VRMs, etc.

This would simplify the BMC daemon's power management, which would only need to request that power be supplied to a given node: Linux takes care of powering up the necessary upstream regulator(s) and refcounting so that power is not disabled when it's still needed.

This also allows the BMC daemon to be restarted/upgraded while the nodes are powered up and in use, since the initialization of the power management GPIOs is no longer the BMC daemon's responsibility.

svenrademakers, 2023-09-30:
Fixed by commit 0afd2a203562165bb3e3d83ad35141e92fc5d712

closed 2023-09-30 by svenrademakers

CFSworks, 2023-09-30:
Worth checking if #53 is resolved by the same commit