turing-machines/BMC-Firmware #7: GPIO LEDs and buttons should perhaps be Linux `gpio-leds` and `gpio-keys` objects?

CFSworks, 2023-04-19:
Linux has very nice LED and event-device subsystems that support LEDs/buttons connected through GPIO pins. The power/reset LEDs/buttons could be added to this subsystem through a straightforward modification to the device tree.

This would simplify the bmc daemon's GPIO management a bit (e.g. no key debouncing code needed there), and also allow users who want to repurpose these I/O elements an easier option to do so.

The BMC reset key can also be tagged `linux,code = <KEY_RESTART>;` so that BMC restarts can be triggered even if the BMC daemon is down for some reason.

svenrademakers, 2023-05-01:
I like the idea, im not sure i want to break any backward compatibility for now. Or is there a way we can keep the current sysfs method in parallel?

CFSworks, 2023-05-02:
It does look like, for robustness, Linux will prevent GPIO access to GPIO pins claimed by LEDs/keys.

I am adding these definitions to my new devicetree with `status = "disabled";` so that they're there and other developers can easily play with it if they want (it will cause the `bmc` daemon not to initialize), but they won't break backwards-compatibility yet.

ruslashev, 2023-06-16:
Fixed in #77, per @svenrademakers

closed 2023-06-16 by ruslashev