turing-machines/BMC-Firmware #9: Device tree configuration mistake breaks USB host support

CFSworks, 2023-04-19:
The OHCI/EHCI devices (needed for BMC to function as a USB host) are failing to bind their drivers, because the `usb1-vbus` regulator's driver can't bind. The kernel log says:

```
[    3.589216] sun8iw20-pinctrl 2000000.pinctrl: pin PB3 already requested by 2501000.uart; cannot claim for 2000000.pinctrl:35
[    3.601842] sun8iw20-pinctrl 2000000.pinctrl: pin-35 (2000000.pinctrl:35) status -22
[    3.610750] reg-fixed-voltage: probe of usb1-vbus failed with error -22
```

The device tree says the vbus regulator can be switched on/off with pin PB3, but on the actual board, PB3 is a serial line to node 3. I don't know how the board itself is wired, but it looks like this line has been moved to PG4?

svenrademakers, 2023-05-01:
PB3 is wrong. As far as I can tell, your hypothesis. let me verify if PG4 is correct

CFSworks, 2023-05-01:
I have since learned that PG4 controls the 5V line in the external USB port, which should actually be *disabled* when the internal USB host is being used.

The `usb1-vbus` regulator should either be removed or replaced with a dummy. No 5V supply is needed to use internal USB.

CFSworks, 2023-05-01:
Also, I am planning on rewriting the devicetree for #5 anyway. This one will be resolved when I do that - could you assign both that and this to me?

PR #18 (2023-05-17): Rewrite the devicetree
This is a full rewrite of the devicetree to start to get things a little more under control, and somewhat document what hardware is in there. With these changes, I notice that Linux boots about 0.2 seconds faster, too!

This is a draft because it is based on my restructured repository. The .dts itself can of course be grabbed and used in the meantime. I have also not yet tested a full image flash; this should not be used seriously until someone can confirm that it doesn't upset U-Boot itself.

I am currently working under the 3-clause BSD license. Is that a suitable FOSS [... cut]
commit 6532b6e3 (2023-05-27): Rewrite the devicetree (#18)

This is a full rewrite of the devicetree to start to get things a little
more under control, and somewhat document what hardware is in there.
With these changes, I notice that Linux boots about 0.2 seconds faster,
too!

This is a draft because it is based on my restructured repository. The
.dts itself can of course be grabbed and used in the meantime. I have
also not yet tested a full image flash; this should not be used
seriously until someone can confirm that it doesn't upset U-Boot itself.

I am currently working under the 3-clause BSD license. Is that a
suitab [... cut]
closed 2023-05-17 by svenrademakers