turing-machines/BMC-Firmware #32: Fixed MAC address

svenrademakers, 2023-05-08:
# Description

_As a user, I want my MAC address not to be reset after a firmware upgrade or reboot_

This issue aims to make a more permanent solution for storing the Mac address by storing it in EEPROM.

# Acceptance criteria

* When booted for the first time, the EEPROM is written with a new MAC address if empty
* The old technique of setting MAC with the tpi.ini file is reverted

CFSworks, 2023-05-09:
Massive :+1: to achieving some type of consistency in the MAC address.

If I could have it my way, I'd want for TMs to get IEEE to assign a proper OUI, so that TMs can allocate real permanent MAC addresses. These should be written to EEPROM at the factory (along with the serial number, board revision, and perhaps other pertinent manufacturing data), with the MAC addresses pulled by the driver so that no userspace intervention is required. For boards already shipped without this EEPROM data, an API on turingpi.com can be set up that allows retrieving it via the serial number -- if the firmware detects a blank EEPROM, it prompts the user for the serial number to hit that API.

For fixed non-permanent MAC addresses (i.e. for the time being), I think we should rely on that very code in U-Boot: I'm currently looking at how difficult it would be to get off of the Allwinner branches and back to vanilla/upstream U-Boot (w/ only a few patches), and as part of that effort I'd like to enable U-Boot's network stack so that it can support TFTP recovery/reflashing. It would be excellent for U-Boot and Linux to agree on the MAC address used, and since U-Boot saves the MAC address in its environment after it is computed, it would be trivial to pass that to Linux in the boot cmdline.

gadkrumholz, 2023-07-15:
I think it should be worth exploring to obtain an OUI for MAC addresses: https://standards.ieee.org/products-programs/regauth/oui/

This will make MAC collisions 100% avoidable. The first 3 of the MAC would be static OUI and the last 3 can be generated from sunxi serial number ( this is customary to do for most SBCs, phones, tablets, etc ). Just my 2 cents.