turing-machines/BMC-Firmware #35: Recovery image

svenrademakers, 2023-05-08:
# Acceptance Criteria 

* an sd card image that automatically flashes the BMC with the latest stable firmware is released. 
* the serial port is used to report progress

CFSworks, 2023-06-06:
Would it be acceptable to combine this with #34? The picture I have in my head is:

- Bootable SDCard with 3 partitions:
  - 16MiB FAT32 "boot" partition, which contains U-Boot env/script files and an initramfs image
  - X MiB SquashFS partition, sized exactly to fit the SquashFS image itself (so determined at build time)
  - X MiB ext4 partition, defaults to a very small size, resized and formatted when we boot in #34 mode.
- There is a marker file in the FAT32 partition called `recovery.txt`, which (when present) makes the bootloader load the initramfs.
- The initramfs contains a single, statically-linked Rust binary that implements both the recovery logic (it can reformat the NAND and write the bootloader and SquashFS to it) and the #57 logic, which makes this image conveniently usable in the factory.

The idea is we ship the recovery .img as part of our standard release process, but the `recovery.txt` contains some text like:
```
The presence of a file named `recovery.txt` in the root of the first partition of the SD card indicates to
the bootloader that we want this to be a "recovery/installer" SD image.
(Technically, it tells the bootloader to load `initramfs.img` instead of doing a normal boot.)

If you would rather use this SD card as your BMC's root+boot drive instead of the builtin flash, delete or
rename this file.

If you change your mind, return this file (or another one; contents are ignored) to the same location on
the SD card.  Changes made to the filesystem while the SD card was used as a root device will not affect
the recovery functionality.
```

svenrademakers, 2023-11-21:
resolved by #126