turing-machines/BMC-Firmware #36: TuringPI Imager

svenrademakers, 2023-05-08:
# Description

_As a user, i want to have a tool I can install on my computer which i can flash the latest bmc firmware with_

# Acceptance Criteria

* There is a Linux and Mac release

jameswlane, 2023-06-04:
I am interested in seeing the priority for a native imager for the BCM due to the fact you are unable to find PhoenixSuit's official page or git repo, and versions and authenticity for most that come up on a standard search are questionable and potentially open you up to malware or a virus at worse. As an interim stop-gap, links to an approved version of the 3rd party utility would also be helpful.

svenrademakers, 2023-06-21:
as i see it i want to have 2 ways to update the firmware.
1. The default way, via the swudate agent, it should be able to download itself the latest firmware and install it. ( we need to update the uboot env a bit so it can A/B rootfs and kernel partitions in case updating went wrong)
2. When the user bricked the device, it can use the "imager tool", it will put the board in fel mode, and writes a complete nand image to it.  There are 2 ways:
    a. the easy way with using the xfel write to write the whole flash.
    b. Write a small Fel tool that loads from RAM so you could use the proprietary driver to write certain partitions etc while being able to perverse bad blocks and write counters) 
    
This ticket is about implementing point 2.

svenrademakers, 2023-10-24:
A lot has happened since the creation of this ticket: 
- We are approaching `v2.0`, meaning the proprietary NAND driver is replaced by a mainline NAND driver. Partitioning and volume management of our flash has changed.
- 'swupdate' is replaced by another component called `os_update,` which can upgrade the firmware on a functioning system.
- an SD card image exists to flash broken or new boards with the latest firmware.

There is still a gap for users who don't have an SD card at hand or cannot access the card slot because the board is mounted in a case. This ticket is meant to fill this gap.

# Requirements

- Must work on boards that have no firmware or firmware in an unknown state
- Architecture must be designed so that the tool can support multiple versions of Turing Machine products.
- Tool works over a USB connection.
- Tool must run on Windows/Linux/Mac.
- GUI is nice to have

# Gathering info

Roughly, the tool will have to execute the following steps:
- boot T113 into FEL
- reformat MTD when a firmware version lower as `v2.0` is detected.
- create new UBI volume
- write firmware image to volume
- signal the bootloader new OS is installed
- (repair or maintenance of existing volumes)

The program will have a 'host' part running on the user's machine. And a 'device' part running on the T113. They will communicate over a USB bus. What logic is executed on which side depends on the chosen communication protocol between the two.  Ideally, we want to keep the device side as simple as possible. This would make the efforts of adding support for more devices easier. 

We use an MTD driver + UBI layer for our eMMC chip. This means that communication is slightly different from conventional block devices. In terms of choosing a USB interface, we can expect some incompatibility. There are no USB interfaces available that can expose MTD volumes, or at least it seems they all expect a block device as input.

# USB interface

## Solution 1A
I investigated if we can expose the flash of the T113 as a Mass storage device to the host. This would mean we mount the block device on the host, do UBI volume management, and write the image inside the tool. Uboot has a command called [`ums`](https://u-boot.readthedocs.io/en/v2021.07/usage/ums.html?highlight=ums), which exactly exposes a block device as MSD.  However, as mentioned earlier, the Mass Storage device class only supports block devices. To work around this, we would need to expose the MTD volume as a block device with the `mtdblock` driver. This would prevent us from mounting the ubi volumes as the ubi tooling cannot work on an MTD device virtualized as a block device. 
**Hence, it seems that this is not a viable solution**

## Solution 1B
Uboot has a second API called [`DFU`](https://u-boot.readthedocs.io/en/v2021.07/usage/dfu.html?highlight=dfu) which does support NAND, We would need to read the whole MTD volume over `DFU`, mount the UBI volumes, do our work and write the whole MTD partition back again.

## Solution 2
We could expose ubi partitions as block devices see `ubiblock` driver. We would still have an USB API to do the volume management on the device side but writing/reading would still happen on the host side over a Mass Storage interface.
This solution implies we need to run a kernel on the device side.

## Solution 3
Implement our own custom USB interface. We move all the logic of volume management and writing of the image to the device. And create an API that the host can call.

I haven't made a conclusive decision yet, but now I prefer solutions 1B and 2. I want to avoid making yet another interface over USB that needs to be future-proof and compatible with other devices.
Solution 1B can work, because our flash chips are not very big in size. Solution 2 would be the most elegant in my opinion.