turing-machines/BMC-Firmware #46: RTL8370MB POC

svenrademakers, 2023-05-18:
# Description

This issue is a tracking issue for all developments regarding controlling the RTL8370MB chip via I2C. As of now, it's unclear if there is a driver available for this chip. As a first task, we want to do an investigation to get a better picture on the challenges and requirements to control this IC.

CFSworks, 2023-05-21:
It looks like the RTL8370MB driver (called "rtl8365mb variant") landed in Linux 5.16. I don't really want to backport the driver because it does seem pretty heavy. (We might want to put this one behind an issue tracking progress for getting on the latest Linux LTS?)

If we're willing to sacrifice access to the EEPROM for the time being, we can reclaim the I²C pins as GPIOs for the realtek-smi driver. But getting these both working at the same time will require a little bit of kernel hacking.

donaldguy, 2024-03-24:
Just throwing in a set of links that ... didn't really help me, but maybe someone else

- https://www.kernel.org/doc/Documentation/networking/dsa/dsa.txt
- https://www.kernel.org/doc/html/next/networking/dsa/configuration.html
- https://netdevconf.info/2.1/papers/distributed-switch-architecture.pdf

... hopefully not where things need to go:
- https://crates.io/crates/mdio 
- possibly a template: https://crates.io/crates/ksz8863