turing-machines/BMC-Firmware #55: Link Local configuration page

svenrademakers, 2023-05-23:
# Description

The idea of this ticket was first mentioned [here](https://github.com/turing-machines/BMC-Firmware/issues/20#issuecomment-1537040796).
We want to set up a link-local connection as a first-time boot or in scenarios where no IP is assigned. The user can edit key settings and configuration via a web page exposed over this link-local connection and thereby enhance user experience.

CFSworks, 2023-05-23:
Would this be a special page that comes up when no DHCP server is detected, or just "make sure HTTPS+mDNS+link local addressing works even if the TP2 board is plugged directly into someone's computer"?

My comment on #20 was just saying that the latter follows naturally from having mDNS up and announcing the current hostname by default.

gjrtimmer, 2025-01-28:
@svenrademakers I might have a very cool idea how to solve this, however I want to check if this has idea has any merrit. For this work I would like to know the following.

If I want to make last minute changes to the filesystem of an image, I need to follow the `post_image.sh` script correct, and place a file in the overlay directory to included in the image as the overlay will be the final version of a file correct?

gjrtimmer, 2025-01-28:
@svenrademakers Thank you, this would work great for me, let me build a prototype give a several days