turing-machines/BMC-Firmware #57: Integrate Factory POST

svenrademakers, 2023-05-23:
At the moment, the factory needs to flash the bmc twice, first to flash the POST test image and afterward the actual firmware. To simplify manufacturing, it would be nice if flashing could be done only once. Ideally, it would boot up in the test environment, when the test is deemed to be successful, the operator can reboot the BMC which it then upgrades itself into an actual firmware image.

CFSworks, 2023-06-05:
I'm interested in making the new "installer SDCard" image also contain diagnostic tests which can be used at the factory before the firmware is flashed (from the same card) and by users wanting to check that their hardware still works correctly. Does inserting a (periodically updated/reimaged) microSD fit into the manufacturing flow or is there some other way those guys initialize the boards?

Can the test image be posted here for the community (but mostly me) to pick apart and see what it does? Or is this considered sensitive? I also want to add several test cases for Ethernet that I think are currently overlooked, as I'd personally received a QC-pass board with bad Ethernet in the past (and I'm a perfectionist when it comes to network equipment).

svenrademakers, 2023-06-06:
my aim is still to have the factory test as part of the standard image. I don't have it hashed out, but I want to automate the firmware so that when the EEPROM is still empty, the firmware(or bootloader) automatically starts the diagnostics test.

that said, i don't know how the test looks and what the exact manufacturing flow is. I will take this discussion up with the hardware guys and take the discussion from there. 

in any way, i think it's an excellent idea to make the test part of the Installer SD. Hopefully, we can catch hardware faults more easily that way when they occur in the field.

svenrademakers, 2023-06-17:
@CFSworks the image used in the factory for testing is located on this branch https://github.com/wenyi0421/turing-pi/tree/testCode

svenrademakers, 2024-07-22:
closing ticket, Our POST factory tests are mainly concerned with testing perhiperals. A very cumbersome procedure for a user to execute. The existence of the fail_safe mode should give the users some extra handles to debug a broken board.

closed 2024-07-22 by svenrademakers