turing-machines/BMC-Firmware #73: Feature Request: support cloud-init when flashing

gclawes, 2023-06-05:
For operating systems that use cloud-init (ubuntu, etc) it would be useful to support injecting cloud-init files into the boot partition.

HypriotOS has a `flash` tool that does this: https://github.com/hypriot/flash

It's a simple matter of mounting the boot partition and copying given `user-data` and `meta-data` files:
https://github.com/hypriot/flash/blob/master/flash#L686-L687
https://github.com/hypriot/flash/blob/master/flash#L704-L712

The `flash` tool also uses this approach for other config file (`config.txt`, etc)

ruslashev, 2023-06-05:
I support the idea, and also wanted something adjacent to this functionality. I think this can be implemented by providing a general command in host CLI to mount node's boot and data partitions in current directory, where afterwards you can copy `user-data` & `meta-data` files, edit config.txt, cmdline.txt etc.

What do you think?

cc @svenrademakers

gclawes, 2023-06-05:
I think being able to pass files in without having to mount/edit manually would help with things like automated provisioning (I'm thinking future features like Kubernetes ClusterAPI providers would benefit from this).

Similar workflow to how OpenStack Ironic's "configuration drive" feature: https://docs.openstack.org/ironic/latest/install/configdrive.html

Another thing to keep in mind is coreos ignition.  It doesn't us `user-data` or `meta-data` like cloud-init, it looks like bare-metal is intended to get configs over the network: https://coreos.github.io/ignition/supported-platforms/

gclawes, 2023-06-20:
Would this be exposing the entire eMMC/SDcard to the modules?  Or just a subdirectory or virtual block device?

If I recall correctly, NBD would only work for read-only if you wanted to support multiple modules at once, for read-write it would have to be NFS if you wanted to support multiple modules accessing it.

I supposed you could expose separate per-module config drives as well for cloud-init/ignition purposes.

j0ju, 2023-07-31:
You could also add the IP addresses 169.254.169.254 or more modern the IPv6 address fe80::a9fe:a9fe on the BMC itself, so the images with cloud init could fetch the data via network from configs prepared to be served via http from the turingPi itself.

gclawes, 2023-07-31:
Yeah, a metadata service on the BMC would be cool