turing-machines/BMC-Firmware #75: documentation request: v1.0.2 upgrade path

mjmenger, 2023-06-06:
The documentation for release v1.0.2 states the following.

```text
To upgrade to 1.0.2 its recommended to flash the .img file using the phoenix suit.
```
To realize the benefits of v1.0.2, is it possible to apply only the .swu file, or is flashing the .img file a required component of the v1.0.2 upgrade?

Thanks

PeterDuckett, 2023-06-07:
I have flashed using the OTA and not had a problem though im a bit stuck as im a linux house hold with no windows machines to use phoenix suit

svenrademakers, 2023-06-08:
Hello @mjmenger!

The .swu files only update the filesystem part of the firmware. Since 1.0.2 has updates inside the kernel, is updating via swupdate not sufficient.

We are working to eliminate the limitations, that the current swu update system imposes on kernel and partition updates. 

using phoenixsuit within a vm is the only current supported setup for linux users.

Let me know if i can help you with setting up the vm or other questions you might have!

closed 2023-06-09 by svenrademakers

mjmenger, 2023-07-06:
> Hello @mjmenger!
> 
> The .swu files only update the filesystem part of the firmware. Since 1.0.2 has updates inside the kernel, is updating via swupdate not sufficient to install a full upgrade to 1.0.2.
> 
> We are working to eliminate the limitations, that the current swu update system imposes on kernel and partition updates. When we successfully did so, we can obsolete the phoenix-suit requirement.
> 
> using phoenixsuit within a vm is the only current supported setup for linux users.
> 
> Let me know if i can help you with setting up the vm or other questions you might have!

Hello @svenrademakers , I would like to take you up on your offer to set up the host for flashing the bmc.

svenrademakers, 2023-07-06:
@ruslashev are you able to share the vm tutorial pages already with @mjmenger? or should we wait until the new readme portal is live?

ruslashev, 2023-07-06:
Yes, I'll update the docs, and address #92 there as well.

ruslashev, 2023-07-07:
Documentation is now live at the [BMC Update](https://docs.turingpi.com/docs/bmc-update) page.