turing-machines/BMC-Firmware #78: Feature: New UI with UART/TTY

PhearZero, 2023-06-09:
I created a UI for the BMC a few weeks ago with xterm. It's very minimal and light weight. Feel free to use it if you like! 

- Mock Demo: https://turing-pi-ui.vercel.app/
- Repo: https://github.com/PhearZero/turing-pi-ui

[image]
[image]

svenrademakers, 2023-06-09:
Cool stuff! We are reconsidering the UI but haven't decided yet on what framework to use or how it will look. we will take this option into consideration when we will get to this point! thanks!

PhearZero, 2023-06-09:
> We are reconsidering the UI but haven't decided yet on what framework to use or how it will look

Technical Details:

- Semantic HTML using [Pico Classless](https://picocss.com/)
- [Sveltekit Framework](https://kit.svelte.dev/)

CFSworks, 2023-06-10:
I would also love this; on the server side, I'd also been looking at using Rust's `vt100` crate to capture (and succinctly replay) the output coming out of the node while the web interface isn't up, so that opening the console in the web UI gets you an up-to-date screen snapshot.

PhearZero, 2023-06-11:
@CFSworks That sounds great, I was using the current UART endpoints. Noticed a lot of basic errors with the REST API and started refactoring it in [my fork](https://github.com/PhearZero/zero-pi-2/blob/issue-5-Feature_V2_REST_API/app/bmc/webserver.c#L925 ). I'm guessing you are planning on dropping goahead 🤞? If it's going to get dropped I'll just hold off until it gets refactored. Might have to finally start using rust more, on a Zig kick right now. 

Tried a bunch of WebTTY projects which where very heavy and ended up with a very crude TTY due to the current UART REST implementation. [share-tty](https://github.com/elisescu/tty-share) is based on gorrilla/GoLang and they are thinking about porting to Rust, we talked a bit about it on a [PR](https://github.com/elisescu/tty-share/pull/68). I really like their protocol design