turing-machines/BMC-Firmware #86: gRPC tpi tool

svenrademakers, 2023-06-24:
# Acceptance Criteria

* the tpi tool builds for windows, mac, linux
* connects to the grpc interface of the bmcd
* current tpi tool is removed
* all functionality not supported by the rpc interface that was in the old tpi tool is moved to the new tool

svenrademakers, 2023-06-26:
created a new repo for this, see: https://www.github.com/turing-machines/tpi

gjrtimmer, 2023-07-26:
@svenrademakers I see that the new TPI tool is written in `rust` is this mandatory? Due to multi arch I was thinking in contributing with golang. Can I give it a try or is `rust` a must?

svenrademakers, 2023-07-27:
Go is probably the best choice, given this tool only does some API calls.  But a strong argument exists to keep everything in the same language/ecosystem. I don't want to make an exception for this case. 

Secondly, I will probably be the one maintaining this, and i wrote like 10 lines of go code in my life. Even though i would not mind learning new things, I want to prevent the potential headaches for now ;) 

At last, I'm almost finished coding this in Rust. you can take a look at: https://github.com/turing-machines/tpi

svenrademakers, 2023-08-07:
`tpi` tool is implemented based on the legacy API. See https://github.com/turing-machines/tpi. The remaining action points are moved to a new ticket: https://github.com/turing-machines/tpi/issues/1

closed 2023-08-07 by svenrademakers