turing-machines/BMC-Firmware #87: Add guide to full-system flashing on Linux hosts via a Windows VM

ruslashev, 2023-06-26:
An unfortunate and not the user-friendliest workaround while we work on a tool for Linux/Mac.

svenrademakers, 2023-07-03:
Could you update the status regarding this here? Thanks!

ruslashev, 2023-07-03:
> @ruslashev Could you update the status regarding this ticket here? Thanks!

I've wrote the draft but the task is put on hold while our docs are being overhauled/put on github to be editable (i.e. a move from HTML-rendered files to GitBook or whatever else team decides upon).

The draft in question is just a brief recommendation to install VirtualBox with a freely-downloadable Windows 11 ISO, since the process is unattended and very straightforward. The only required next step is to add a catch-all filter of USB devices and the rest of the steps are already described in the existing guide.

svenrademakers, 2023-07-03:
Thanks for the update, will move it to the next milestone then

ruslashev, 2023-07-07:
Done in the [BMC Update](https://docs.turingpi.com/docs/bmc-update) page.

closed 2023-07-19 by ruslashev