turing-machines/BMC-Firmware #104: Firmware version is not propagated

svenrademakers, 2023-07-28:
**Describe the bug**
As a user, I want to be able to print the version of the turingpi firmware so I can see what version is installed. 
Im pretty sure this was working before, but now the firmware version is not showing when running `uname -a`

**To Reproduce**
* Flash v1.1.0
* run `uname -r`
* version shows `5.4.61`

**Expected behavior**
version to show `5.4.61-turingpi-1.1.0`

**Versions**
linux version= 5.4.61
bmc version= 1.1.0

CFSworks, 2023-07-28:
What's our stance on `/etc/os-release`?

walkjivefly, 2023-07-28:
If it helps, the CE firmware says
```
# uname -a
Linux turing 5.4.61 #31 SMP PREEMPT Thu Oct 20 00:11:14 CST 2022 armv7l GNU/Linux
# uname -r
5.4.61
```

j0ju, 2023-07-28:
uname shows the version of the operationg system only if compiled into the kernel. If the kernel is not compiled and updated by the .swu this is very unlikly.

cprima, 2023-08-18:
> What's our stance on `/etc/os-release`?

Thanks for mentioning this source!
I wrote me a sh script https://gist.github.com/cprima/e09a5d2f4151f38cc98f2b2a196a0ee1 to initialize nodes, and could replace a curl request to the API with a simple parser of that file.

svenrademakers, 2023-09-27:
2 things are important:
* it should be possible to trace back a given build
* this version should be reproducible (deterministic). 

furthermore i advocate to keep buildroot as vanilla as possible. 
`/etc/os-release` gives us the info needed to reproduce a given build. Hence no additional work to be done here.

closed 2023-09-27 by svenrademakers