turing-machines/BMC-Firmware #105: Release 1.1.0 is missing "*.swu" asset

gokhanbarisaker, 2023-07-29:
**Describe the bug**
As a user, I would like to download a pre-built swu file to be able to upgrade

**To Reproduce**
Steps to reproduce the behavior:
1. Go to https://github.com/turing-machines/BMC-Firmware/releases/tag/1.1.0
2. Scroll down to 'Assets'
3. Observe that the swu file is not among the uploaded assets

**Expected behavior**
A downloadable swu file to be available under each release

**Screenshots**
[image]

**Versions**
-

**Additional context**
-

svenrademakers, 2023-07-29:
The missing .swu file is intentional. Providing kernel updates inside the .swu package could brick your device if you don't have a Windows machine to rewrite the flash on your board. The decision is therefore made to be conservative and not distribute a .swu asset for this release. We expect to have an alternative solution in place in the near future. tracking #35 #36.

gokhanbarisaker, 2023-07-29:
Makes sense. Would it be possible the highlight this situation on either the release page or the documentation? As someone with not enough knowledge, it feels like the upgrade is not possible until those issues are resolved

svenrademakers, 2023-07-29:
This is something we should do asap. Thanks for the input! we will be updating the docs in the course of next week 👍

suranyami, 2023-07-31:
Is there reference information anywhere as to what the Windows-specific software/hardware is that is needed to re-flash if BMC is bricked?

I'd be keen to have a go at porting it to linux/macOS if it's just software.

svenrademakers, 2023-07-31:
The Windows tool in question is called [PhoenixSuit](https://docs.turingpi.com/docs/turing-pi2-bmc-update#flashing-on-a-windows-host). There are multiple hoops you need to jump through if you were to attempt such thing.
* The generated .img is fully proprietary. You would need to spend a lot of time reverse-engineering the format or do a new thing, which implies a major overhaul of the build script and tools. 
* Secondly, the flash driver on the board is also proprietary, you would need to work around the SIMULATE_MULTIPLANE mode which is enabled. 

I would strongly discourage you not do it. Not only because of the effort involved, but also because we are well underway of doing some major refactoring here to move away from all this proprietary stuff.

svenrademakers, 2023-08-07:
Updated the release notes with a small explanation of the missing .swu asset. Further developments with regards to firmware update will be tracked with #36.

closed 2023-08-07 by svenrademakers