turing-machines/BMC-Firmware #106: Unable to build the docker image on macos ventura (13.4) with docker desktop 4.21.1 (114176)

gokhanbarisaker, 2023-07-29:
**Describe the bug**
As a user, I would like to be able to build provided docker file. So that, I could generate my own assets.

I am unable to build the docker image due to missing `lib32z1` package on macos ventura (13.4) with docker desktop 4.21.1 (114176)

**To Reproduce**
Steps to reproduce the behavior:
1. Run 'docker build .'
4. See error (RUN command has been partitioned into individual commands with the below example to be able to have a clear demonstration.)

```
 => ERROR [22/29] RUN apt-get install -y lib32z1                                                                    0.8s
------
 > [22/29] RUN apt-get install -y lib32z1:
0.295 Reading package lists...
0.684 Building dependency tree...
0.763 Reading state information...
0.769 E: Unable to locate package lib32z1
------
Dockerfile:24
--------------------
  22 |     RUN apt-get install -y cvs
  23 |     RUN apt-get install -y unzip
  24 | >>> RUN apt-get install -y lib32z1
  25 |     RUN apt-get install -y lib32z1-dev
  26 |     RUN apt-get install -y lib32stdc++6
--------------------
ERROR: failed to solve: process "/bin/sh -c apt-get install -y lib32z1" did not complete successfully: exit code: 100
```

**Expected behavior**
Docker image to be built successfully

**Screenshots**
-

**Versions**
-

**Additional context**
-

svenrademakers, 2023-07-31:
I have a huge that one of the mirrors was available intermittently.
Are you able to post your complete docker log?
Is your host on Apple silicon?

gokhanbarisaker, 2023-08-05:
This is the console output that I got with the `docker build --no-cache  --progress=plain .` command on the latest master branch (https://github.com/turing-machines/BMC-Firmware/tree/57409d5a30d4b63e60b6b13007a3ce371250bff6).

```
#0 building with "desktop-linux" instance using docker driver

#1 [internal] load .dockerignore
#1 transferring context: 2B done
#1 DONE 0.0s

#2 [internal] load build definition from Dockerfile
#2 transferring dockerfile: 511B done
#2 DONE 0.0s

#3 [internal] load metadata for mcr.microsoft.com/devcontainers/base:ubuntu
#3 DONE 0.0s

#4 [1/2] FROM mcr.microsoft.com/devcontainers/base:ubuntu@sha256:fdb79b61fb4e955b4042ddb25bbae616316984bcd46c193196c6e1664a5d367f
#4 CACHED

#5 [2/2] RUN apt-get update && apt-get install -y   build-essential   subversion   git-core   libncurses5-dev   zlib1g-dev   gawk   bc   cpio   file   flex   quilt   libssl-dev   xsltproc   libxml-parser-perl   mercurial   bzr   ecj   cvs   unzip   lib32z1   lib32z1-dev   lib32stdc++6   libstdc++6   libncurses-dev   u-boot-tools   mkbootimg   && rm -rf /var/lib/apt/lists/*
#5 0.155 Get:1 http://ports.ubuntu.com/ubuntu-ports jammy InRelease [270 kB]
#5 0.269 Get:2 http://ports.ubuntu.com/ubuntu-ports jammy-updates InRelease [119 kB]
#5 0.306 Get:3 http://ports.ubuntu.com/ubuntu-ports jammy-backports InRelease [109 kB]
#5 0.350 Get:4 http://ports.ubuntu.com/ubuntu-ports jammy-security InRelease [110 kB]
#5 0.382 Get:5 http://ports.ubuntu.com/ubuntu-ports jammy/main arm64 Packages [1758 kB]
#5 0.480 Get:6 http://ports.ubuntu.com/ubuntu-ports jammy/restricted arm64 Packages [24.2 kB]
#5 0.480 Get:7 http://ports.ubuntu.com/ubuntu-ports jammy/universe arm64 Packages [17.2 MB]
#5 0.904 Get:8 http://ports.ubuntu.com/ubuntu-ports jammy/multiverse arm64 Packages [224 kB]
#5 0.907 Get:9 http://ports.ubuntu.com/ubuntu-ports jammy-updates/restricted arm64 Packages [518 kB]
#5 0.918 Get:10 http://ports.ubuntu.com/ubuntu-ports jammy-updates/universe arm64 Packages [1117 kB]
#5 0.941 Get:11 http://ports.ubuntu.com/ubuntu-ports jammy-updates/multiverse arm64 Packages [27.8 kB]
#5 0.984 Get:12 http://ports.ubuntu.com/ubuntu-ports jammy-updates/main arm64 Packages [993 kB]
#5 1.178 Get:13 http://ports.ubuntu.com/ubuntu-ports jammy-backports/main arm64 Packages [48.8 kB]
#5 1.178 Get:14 http://ports.ubuntu.com/ubuntu-ports jammy-backports/universe arm64 Packages [23.7 kB]
#5 1.178 Get:15 http://ports.ubuntu.com/ubuntu-ports jammy-security/multiverse arm64 Packages [23.4 kB]
#5 1.178 Get:16 http://ports.ubuntu.com/ubuntu-ports jammy-security/restricted arm64 Packages [517 kB]
#5 1.189 Get:17 http://ports.ubuntu.com/ubuntu-ports jammy-security/main arm64 Packages [730 kB]
#5 1.210 Get:18 http://ports.ubuntu.com/ubuntu-ports jammy-security/universe arm64 Packages [866 kB]
#5 1.620 Fetched 24.7 MB in 2s (16.3 MB/s)
#5 1.620 Reading package lists...
#5 2.024 Reading package lists...
#5 2.397 Building dependency tree...
#5 2.474 Reading state information...
#5 2.498 E: Unable to locate package lib32z1
#5 2.498 E: Unable to locate package lib32z1-dev
#5 2.498 E: Unable to locate package lib32stdc++6
#5 2.498 E: Couldn't find any package by regex 'lib32stdc++6'
#5 ERROR: process "/bin/sh -c apt-get update && apt-get install -y   build-essential   subversion   git-core   libncurses5-dev   zlib1g-dev   gawk   bc   cpio   file   flex   quilt   libssl-dev   xsltproc   libxml-parser-perl   mercurial   bzr   ecj   cvs   unzip   lib32z1   lib32z1-dev   lib32stdc++6   libstdc++6   libncurses-dev   u-boot-tools   mkbootimg   && rm -rf /var/lib/apt/lists/*" did not complete successfully: exit code: 100
------
 > [2/2] RUN apt-get update && apt-get install -y   build-essential   subversion   git-core   libncurses5-dev   zlib1g-dev   gawk   bc   cpio   file   flex   quilt   libssl-dev   xsltproc   libxml-parser-perl   mercurial   bzr   ecj   cvs   unzip   lib32z1   lib32z1-dev   lib32stdc++6   libstdc++6   libncurses-dev   u-boot-tools   mkbootimg   && rm -rf /var/lib/apt/lists/*:
1.178 Get:16 http://ports.ubuntu.com/ubuntu-ports jammy-security/restricted arm64 Packages [517 kB]
1.189 Get:17 http://ports.ubuntu.com/ubuntu-ports jammy-security/main arm64 Packages [730 kB]
1.210 Get:18 http://ports.ubuntu.com/ubuntu-ports jammy-security/universe arm64 Packages [866 kB]

2.397 Building dependency tree...
2.474 Reading state information...
2.498 E: Unable to locate package lib32z1
2.498 E: Unable to locate package lib32z1-dev
2.498 E: Unable to locate package lib32stdc++6
2.498 E: Couldn't find any package by regex 'lib32stdc++6'
------
Dockerfile:3
--------------------
   2 |
   3 | >>> RUN apt-get update && apt-get install -y \
   4 | >>>   build-essential \
   5 | >>>   subversion \
   6 | >>>   git-core \
   7 | >>>   libncurses5-dev \
   8 | >>>   zlib1g-dev \
   9 | >>>   gawk \
  10 | >>>   bc \
  11 | >>>   cpio \
  12 | >>>   file \
  13 | >>>   flex \
  14 | >>>   quilt \
  15 | >>>   libssl-dev \
  16 | >>>   xsltproc \
  17 | >>>   libxml-parser-perl \
  18 | >>>   mercurial \
  19 | >>>   bzr \
  20 | >>>   ecj \
  21 | >>>   cvs \
  22 | >>>   unzip \
  23 | >>>   lib32z1 \
  24 | >>>   lib32z1-dev \
  25 | >>>   lib32stdc++6 \
  26 | >>>   libstdc++6 \
  27 | >>>   libncurses-dev \
  28 | >>>   u-boot-tools \
  29 | >>>   mkbootimg \
  30 | >>>   && rm -rf /var/lib/apt/lists/*
  31 |
--------------------
ERROR: failed to solve: process "/bin/sh -c apt-get update && apt-get install -y   build-essential   subversion   git-core   libncurses5-dev   zlib1g-dev   gawk   bc   cpio   file   flex   quilt   libssl-dev   xsltproc   libxml-parser-perl   mercurial   bzr   ecj   cvs   unzip   lib32z1   lib32z1-dev   lib32stdc++6   libstdc++6   libncurses-dev   u-boot-tools   mkbootimg   && rm -rf /var/lib/apt/lists/*" did not complete successfully: exit code: 100
```

mcam10, 2023-11-11:
@gokhanbarisaker  Hi,  Not sure if you're still having issues building this.. but I was able to build this with a few changes in my Dockerfile..  

removing the following:
`lib32z1 `
`lib32z1-dev`
`lib32stdc++6`
`libncurses-dev `

and adding the following:
`libz1`

@svenrademakers Not sure if this is the recommended approach, but in case you were curious about the host specs... Im using MacOS Ventura 13.5.1 and using the M2 Pro chip.

Full Dockerfile below:

```
FROM mcr.microsoft.com/devcontainers/base:ubuntu

RUN apt-get update && apt-get install -y \
  build-essential \
  subversion \
  git-core \
  libncurses5-dev \
  zlib1g-dev \
  gawk \
  bc \
  cpio \
  file \
  flex \
  quilt \
  libssl-dev \
  xsltproc \
  libxml-parser-perl \
  mercurial \
  bzr \
  ecj \
  cvs \
  unzip \
  libz1 \
  libstdc++6 \
  u-boot-tools \
  mkbootimg \
  && rm -rf /var/lib/apt/lists/*

j0ju, 2023-11-15:
I can confirm @mcam10 solution works on my M1/aarch64 system.

svenrademakers, 2023-11-15:
First of all, i think its a good idea to start using the `Dockerfile` on CI builds. This way, problems will become more visible/ urgent.
As i understand from your investigations, basically the Microsoft/devcontainer is a moving target, and at some point ubuntu stopped hosting the `lib32` variants because multitarget is the defacto way. A second action point is to pin down the ubuntu base layer.

I need help understanding how the host architecture influences the actual container. Mainly because on MacOs, you just get a full-fledged virtual machine shoved under your container.

j0ju, 2023-11-16:
I was able to build with the following patch (below). I just tested on AMD64, test runs on Docker via M1 and and CM4 node will be done in the next days when I have time.
I have tested with this three command in the docker container, I did not test the generation of the .swu file (isn't that obsolete anyways?)
```shell
cd buildroot
make BR2_EXTERNAL=../tp2bmc tp2bmc_defconfig
make
```
I also patched the readme.md slightly, to enter the container , if you don't have standard docker install on a linux workstation (Debian/Ubuntu), then the build data generated during the build will be owned by root outside of the container. That is a bit annoying and might confuse some users. This is not an issue the way it works on Docker OSX or Rancher Desktop.
> docker run -it -v $PWD:$PWD --workdir $PWD -u $UID:$GID tp_container
For OSX users it might be convenient to use Docker volumes to speed up the builds significantly. This is due to way the filesystem is exposed into the container with `-v` on OSX.

Patch:
```diff
diff --git a/Dockerfile b/Dockerfile
index 44798889..a612d44e 100644
--- a/Dockerfile
+++ b/Dockerfile
@@ -8,7 +8,7 @@ RUN apt-get update && apt-get install -y \
   zlib1g-dev \
   gawk \
   bc \
-  cpio \ 
+  cpio \
   file \
   flex \
   quilt \
@@ -20,9 +20,7 @@ RUN apt-get update && apt-get install -y \
   ecj \
   cvs \
   unzip \
-  lib32z1 \
-  lib32z1-dev \
-  lib32stdc++6 \
+  zlib1g-dev \
   libstdc++6 \
   libncurses-dev \
   u-boot-tools \
diff --git a/readme.md b/readme.md
index 4bbc6336..fed2fc91 100644
--- a/readme.md
+++ b/readme.md
@@ -53,7 +53,7 @@ of your repository:
 docker build . -t tp_container
 
 # open the docker container
-docker run -it -v $PWD:$PWD --workdir $PWD tp_container
+docker run -it -v $PWD:$PWD --workdir $PWD -u $UID:$GID tp_container
 ```
 
 ### Natively
 ```

mcam10, 2023-11-16:
Nice @j0ju. I can work on the Dockerfile integration on the CI builds.. if no-one else is taking that on? @svenrademakers

j0ju, 2023-11-16:
I tested this now on an amd64 and on arm64 host and on an M1 Pro with docker desktop.
It works building the image. Not yet tested, but I don't see issues there as Buildroot is building needed tools for the image generation on itself. If that works and I assume this gets rid of i386/amd64 depedency on building.

I would also update the `readme.md` (and uppercase it's name, as it triggers my inner monk ;) )
Adding a `.dockerignore' file with contents 
```
buildroot
```
would prevent the inclusion of the entire buildroot directory on container rebuilds, which might be large.

The manual containerized build process for linuxoid hosts would be best with:
```
# create container
docker build . -t tp_container

# enter container
docker run -it -v $PWD:$PWD/buildroot --workdir $PWD -u $UID:$GID tp_container

# build
make BR2_EXTERNAL=../tp2bmc tp2bmc_defconfig
make
```

At least for Rancher Desktop building with the aboves method does not work, as the filesystem is then shared via sshfs, with has some limitations. I cannot say for Docker Desktop.
For builds on OSX machines independently if (Intel or M1 or Rancher Desktop or Docker Desktop) would be:
```
# create container
docker build . -t tp_container

# create a native volume
docker volume create turing-pi-bmc-firmware

# enter the container
docker run -it -v turing-pi-bmc-firmware:/src -w /src tp_container

# check the repository out again on the volume inside of the docker VM
git clone https://github.com/turing-machines/BMC-Firmware.git
cd BMC-Firmware/buildroot
 
# build
make BR2_EXTERNAL=../tp2bmc tp2bmc_defconfig
make
```
Notes: the resulting images are then on the volume not on the native filesystem of OSX, and needs to be copied out befor using.

svenrademakers, 2023-11-16:
@mcam10 ah! I missed your message could have spared me some time! 

I updated the Github CI to build and use the Dockerfile. I only did not yet created a virtual disk to checkout the repo in.

I will open a PR with what i have right now. Perhaps you guys can make some additions if applicable, i like what i read so far

svenrademakers, 2023-11-16:
see https://github.com/turing-machines/BMC-Firmware/pull/142

j0ju, 2023-11-16:
Depending how fast you want do merge it, I can have a look if the README.md changes are in and add missing and add notes.
Just to clarify I would not use the docker volume for CI pipeline runs: I would only document it for those that are building manually on OSX hosts.

mcam10, 2023-11-16:
Nice @svenrademakers... all good! Just trying to help anyway I can. I guess an addition to this would be locking down the base image version in the Dockerfile.

So `mcr.microsoft.com/devcontainers/base:ubuntu22.04`

hoinzy, 2023-11-17:
I want to briefly mention here what probably happened on the M1 Mac:

The processors can emulate x86_64, but not the i386 instruction set. And I guess the virtual machine sitting below docker also doesn't emulate those instructions either.

svenrademakers, 2023-11-17:
@j0ju i left the honors of updating the Docker section in the README.md to you :)

commit d1670a00 (2023-11-16): CI: build using docker image

By using docker container in the CI, we create more visibility on the
state of the docker image. Any docker failures will result in a
failed CI build.

fixes #106. Dockerfile definition was using outdated definitions of
packages.
commit 83385949 (2023-11-16): CI: build using docker image

By using docker container in the CI, we create more visibility on the
state of the docker image. Any docker failures will result in a
failed CI build.

fixes #106. Dockerfile definition was using outdated definitions of
packages.
commit c96ca06b (2023-11-16): CI: build using docker image

By using docker container in the CI, we create more visibility on the
state of the docker image. Any docker failures will result in a
failed CI build.

fixes #106. Dockerfile definition was using outdated definitions of
packages.
commit 3c3becba (2023-11-16): CI: build using docker image

By using docker container in the CI, we create more visibility on the
state of the docker image. Any docker failures will result in a
failed CI build.

fixes #106. Dockerfile definition was using outdated definitions of
packages.
closed 2023-11-17 by svenrademakers