turing-machines/BMC-Firmware #107: Web interface and TPI not responsive

jsilva-iii, 2023-08-01:
Firmware :  1.1.0

Upgraded BMC to 1.1.0 - after a few reboots, the web interface is not responsive, - this is the screen output after a fresh reboot :
start webserver -
************************************************************************************************************************
* [libgo]       rel versiion V1.0.0.7  svn version: Unversioned directory   build time: 2022-11-14 23:27:17
************************************************************************************************************************
start uart thread node num = 3----
goahead: 2: Configuration for Embedthis GoAhead
goahead: 2: ---------------------------------------------
goahead: 2: Version:            3.6.5
goahead: 2: BuildType:          Debug
goahead: 2: CPU:                arm
goahead: 2: OS:                 linux
goahead: 2: Host:               127.0.0.1
goahead: 2: Directory:          /root
goahead: 2: Documents:          /mnt/var/www
goahead: 2: Configure:          me -d -q -platform linux-x86-default -configure . -gen make
goahead: 2: ---------------------------------------------
goahead: 2: Started http://*:80
======= ASP_Init =====func_num 3=====
ASP_Init jst : name app_WebS_Jst_test
 ASP_Init act : name app_WebS_Act_test
 ASP_Init act : name bmc
*** stack smashing detected ***: terminated

**Versions**
linux version= 5.4.61
bmc version=  1.1.0

I looked in /var/logmessages and saw nothing as well an in dmesg.

I re-flashed the firmware and it is ok

gadkrumholz, 2023-08-05:
same here, flashed 1.1.0 using PhoenixSuit and worked for a bit... a few reboots later started getting this same error with `bmc` executable

svenrademakers, 2023-09-15:
fixed by https://github.com/turing-machines/bmcd/commit/96893184e339b0adf50b15de8e44c155abdb3b95

closed 2023-09-15 by svenrademakers