turing-machines/BMC-Firmware #111: Feature Request: Secure web interface

iamwillbar, 2023-08-07:
**Is your feature request related to a problem? Please describe.**
The web interface for the BMC allows privileged operations, such as over-the-air flashing and node power on/off, as such it should have some way to protect from unauthorized users.

**Describe the solution you'd like**
The web interface should be exposed via TLS and should support HTTP basic authentication.

**Describe alternatives you've considered**
Using firewall to block access to web interface, however, this can't protect from nodes on the same subnet.

**Additional context**
Most BMCs support this capability.

svenrademakers, 2023-08-07:
Thanks for submitting this feature request. Currently we are working on a TLS implementation which is tracked in the `tpi_rs` soon to be `bmcd` github repo. see https://github.com/turing-machines/tpi_rs/issues/23.
If there are no further objections i will close this ticket as duplicate

iamwillbar, 2023-08-07:
@svenrademakers will that include the ability to lock down access to the web interface? Even basic auth would be sufficient (once there is TLS). If that's covered by that other issue then no concerns with closing this one, if it's not, I can edit this one to just cover the authentication piece?

svenrademakers, 2023-08-08:
I processed your comments and updated the tickets accordingly.

Related tickets are as follows:

Client-side:
- https://github.com/turing-machines/tpi/issues/2
- https://github.com/turing-machines/tpi/issues/3

Server-side:
- https://github.com/turing-machines/tpi_rs/issues/28

Tls is part of https://github.com/turing-machines/tpi_rs/issues/23.

Expect this work to be picked up after https://github.com/turing-machines/tpi_rs/issues/23 is completed

closed 2023-08-08 by svenrademakers