turing-machines/BMC-Firmware #113: Documentation improvement:

cprima, 2023-08-15:
**Is your feature request related to a problem? Please describe.**
For a Raspberry Pi Compute Module 4, the flashing via command line requires no special cable or special software. The documentation should reflect this 

**Describe the solution you'd like**
Below is a (verbose) markdown text describing the  "Flashing from a command line"

**Describe alternatives you've considered**
If requested, I could also make an animated gif -- but the taste in color scheme varies so I did refrain from making one in the first place.

**Additional context**
As a reader of the documentation I also do not understand the relevance of the informational dic "Versions ≤ 1.1.0 of the firmware contain an outdated web server…" I believe it should be removed from the section "Flashing from a command line" as (for a newbie user like me) how webserver and commandline are related is not immediately clear.

I also do not understand (as a newbie user) the significance of "**(Temporary, only for v1.1.0)** Upload the file to your microSD card" (emphasis mine). Did the SD card version not work with lower BMC versions? Maybe the "(Temporary, only for v1.1.0)" can be removed?

Here my suggested update:

###  As-is documentation:

1. Acquire the desired OS image.
This guide uses Raspberry Pi OS, which you can download from the [official website](https://www.raspberrypi.com/software/operating-systems/). After downloading, extract the image to your filesystem.
1. (Temporary, only for v1.1.0) Upload the file to your microSD card:
`scp 2023-05-03-raspios-bullseye-arm64-lite.img root@turingpi.local:/mnt/sdcard/images`
1. Turn on the desired node through a web interface or a command line.
1. Log in to the BMC through SSH:
1. Perform flashing on desired node.
`tpi --node 1 -l -f /mnt/sdcard/images/raspios.img`
1. (Optional) Access the node's filesystem:

```sh

tpi -n 1 -m
mount /dev/sda1 /mnt/bootfs # Alternatively, you can find it in /dev/disk/by-label/
vi /mnt/bootfs/config.txt # add enable_uart=1 to [all] to enable serial interconnection
umount /mnt/bootfs
tpi -n 1 -x
```
### suggested update

1. Acquire the desired OS image.
This guide uses Raspberry Pi OS, which you can download from the [official website](https://www.raspberrypi.com/software/operating-systems/). After downloading, extract the image to your filesystem.
1. (Temporary, only for v1.1.0) Insert a SD card into the bac of the Turing Pi 2 board. Upload the file to your microSD card:
`scp 2023-05-03-raspios-bullseye-arm64-lite.img root@turingpi.local:/mnt/sdcard/images`
1. Log in to the BMC through SSH:
```
$ ssh root@turingpi.local
```
On first connection you get
```
The authenticity of host 'turingpi.local (10.x.y.z)' can't be established.
ECDSA key fingerprint is SHA256:ABCD1234EFGH5678IJKL90MNOPQRstuvWXYZabcd.
Are you sure you want to continue connecting (yes/no/[fingerprint])?
```
Type `yes`. You will be asked for the password, the default password is turing
```
root@turingpi.local's password:
```
4. Perform flashing on desired node.
Issue the following command which from the local filesystem flashes the image (if the node was powered off this will power on the node (at least as of v1.1.0)):
`# tpi -n 1 -l -f /mnt/sdcard/images/2023-05-03-raspios-bullseye-armhf-lite.img`
When the LEDs next to the node (from the back panel side) is constantly red the power is on.
While the command is running 
- on the back of the carrier board both LEDs will be constantly green (the upper for power, the lower for activity)
- you will get the following output (the 00:00:53 will be counting up, typically between 5 and 11 minutes):
```
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:--  0:00:53 --:--:--     0
```
Only the column "Time Spent" is changing, other columns remain 0. This is standard behavior.
When the command is finished
  - on the back of the carrier board the lower LED typically blinks occasionally green and
  - the output looks like this:
```
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100    56    0    56    0     0      0      0 --:--:--  0:10:16 --:--:--    12
```
5. (Optional) Access the node's filesystem:
Use tpi to load the node as mass storage device:
```
# tpi -n 1 -m
```
While running the output will be:
```
booting node 0 into msd mode. please wait a moment..
```
The command has finished when the output looks like:
```
booting node 0 into msd mode. please wait a moment..
{"response":[{"result":"ok"}]}{"response":[{"result":"ok"}]}#
```
You can now mount the partitions as device, the boot partition is typically `/dev/sda1`. If not, then `dmesg` should return the device name in the last line(s) of its output.
```
# mount /dev/sda1 /mnt/bootfs
```
You can now configure e.g. in Raspbian's boot partition the file config.txt. To enable uart messages on the node's Raspbian (to fetch with `tpi --uart=get -n 1`) add or enable the line `enable_uart=1`:
```
# vi /mnt/bootfs/config.txt
```
Depending on your OS there are other changes that you might want to make. If you are flashing DietPi you will want to edit dietpi.txt.
Once finished, leave vi with `:q<enter>` ;) and unmount the filesytem:
```
umount /mnt/bootfs
```
6. Reboot the node. while running
  - on the back of the carrier board the lower LED will show node activity
```
# tpi -n 1 -x
```
The output will be:
```
{"response":[{"result":"ok"}]}{"response":[{"result":"ok"}]}{"response":[{"result":"ok"}]}#
```
Check if the node did begin to boot by querying the uart output (no matter the above setting in config.txt a few lines of output should be returned once):
```
# tpi --uart=get -n 1
{
        "response":     [{
                "uart": "OTP boardrev d03141 bootrom 48b0 48b0\r\n\r\nRPi: RPIBOOT release VERSION:8ba17717 DATE: 2023/01/11 TIME: 17:40:52\r\nBOOTMODE: 0x07 partition 0 build-ts BUILD_TIMESTAMP=1673458852 serial 5467c368 boardrev d03141 stc 4042127\r\nPM_RSTS: 0x00001000\r\npart 00000000 reset_info 00000000\r\nuSD voltage 3.3V\r\nInitialising SDRAM 'Micron' 32Gb x2 total-size: 64 Gbit 3200\r\nDDR 3200 1 0 64 152\r\nBoot mode: RPIBOOT (03) order f1\r\nRead config.txt bytes        0 hnd 0x0\r\nSIG pieeprom.sig 0000000000000000000000000000000000000000000000000000000000000000 0\r\nBad signature pieeprom.sig\r\nRead start4.elf bytes   301628 hnd 0x0\r\nRead fixup4.dat bytes        0 hnd 0x0\r\n0x00d03141 0x00000000 0x00000fff\r\nMEM GPU: 76 ARM: 948 TOTAL: 1024\r\nStarting start4.elf @ 0xc1000200 partition -1\r\n+\r\n\r"
            }]
```
If the node does not boot try to remedy this by setting the node back to USB host mode (in case `tpi -n 1 -m` did set the node in device mode, seems to be the case for the last inserted module):
```
# tpi -u host -n 
```
The output will be:
```
# tpi -n 1 -u host
{"response":[{"result":"ok"}]}#
```
If the node still does not boot toggle power off and on:
```
# tpi -p off -n 1
# tpi -p on -n 1
```
With the node booted you should be able to find its IP. In case of a home network your router's DHCP might show the IP for the host `raspberrypi `.

svenrademakers, 2023-08-15:
Thanks for the feedback! I tried to incorporate it in our docs. Could you review this once more to see if it's to your liking?
see
https://docs.turingpi.com/docs/raspberry-pi-cm4-install-os-using-turing-pi2-cluster

cprima, 2023-08-15:
Oh, what a quick turnaround time! Many thanks for accepting my change request!

As a proofreading exercise, I just copy&pasted every command, and they work on my system.

I saw you found a neat styling with the roman numerals as the 2nd level ordered list. I also liked the change in the blue (i)nfo div, even the newbie that is me now understands! :)

Could you maybe change this, it should be clearer for the reader:

### Change A

As is:
7. Reboot the node. while running
on the back of the carrier board, the lower LED will show node activity
```
# tpi -n 1 -x
```

To be:
7. Reboot the node:
```
# tpi -n 1 -x
```
While this command is running
i) on the back of the carrier board the lower LED will show node activity

### Change B

as is:
leave vi with :q<enter>

to be
leave vi with :wq<enter>

### Change C

As is:
in case tpi -n 1 -m did set the node in device mode, seems to be the case for the last inserted module

To be:
in case tpi -n 1 -m did set the node in device mode, seems to be the case for the last inserted, powered module

---
Next weekend I might find time to make a screenrecording, e.g. with aciinema, which I should be able to convert into an animated gif.
Only last weekend I booted my Kickstarter Turing Pi 2 for the first time. It is during this initial exploration that I take my time and document what I find helpful.

cprima, 2023-08-15:
Here is an asciinema recording of a completely headless Raspbian installation:
https://asciinema.org/a/602865?t=10:33
As asciinema cannot detect minor changes it recorded boring 10 minutes, over the weekend I might find a way to cut those out.

svenrademakers, 2023-08-18:
Thank you very much for these valuable additions. The community will appreciate it. I updated the changes accordingly. If there are still additions, feel free to open a new ticket

closed 2023-08-18 by svenrademakers

cprima, 2023-08-24:
Hope that this shameless plug of mine does not re-open the issue, but I wanted to mention that my 
script `init-nodes.sh`
available at [https://gist.github.com/cprima/e09a5d2f4151f38cc98f2b2a196a0ee1](https://gist.github.com/cprima/e09a5d2f4151f38cc98f2b2a196a0ee1)
is intended to make an CLI installation much easier, here a sample command to flash 2 CM4 in slot 2 and 3:
```
./init-nodes.sh 2..3 node 2023-05-03-raspios-bullseye-armhf-lite.img
```
I maintain a README in that gist with further instructions.