turing-machines/BMC-Firmware #114: Ability to format bmc SD card with supported filesystems

meierra, 2023-08-16:
I recently found that the SD card installed in my TuringPi2 was not formatted with a supported filesystem, I was able to use ssh and fdisk to create the partition, but no valid mkfs for the supported filesystem types was available, (I could only find mkfs.ubifs)

Would it be possible to include mkfs for the supported sdcard formats (exFAT, NTFS, FAT32, EXT4) and later possibly the ability to format the SDcard via the UI?

svenrademakers, 2023-09-21:
I'm sorry for not getting back to you sooner. Your point is valid. We are tight regarding flash space, but this seems like low-hanging fruit.

meierra, 2023-09-21:
Hi Sven,

All good on the timing, thanks for looking at this

Robert

thz, 2023-10-06:
Also, having a mkfs.vfat would enable creation of /boot filesystem for rpi systems. This would allow installation of custom rpi systems, where you copy contents of /boot over the network (scp) instead of flushing the image.

PR #129 (2023-10-10): configs/br2: add ext2-4 and FAT filesystem tools
This adds `mkfs` and `fsck` tools for ext2-4 and FAT filesystems.

Binaries:

- `mkfs.{msdos,vfat}`, `mkdosfs` -> `mkfs.fat`
- `mkfs.ext{2,3,4}` -> `mke2fs`
- `fsck.{msdos,vfat}`, `dosfsck` -> `fsck.fat`
- `fsck.ext{2,3,4}` -> `e2fsck`
- `mklost+found`

Flash size change:

|   | B | MiB |
|--------|--------|--------|
| before | 60506784 | 57.70 |
| after | 61727251 | 58.87 |
| delta | 1220467 | 1.16 | 

Closes #114
closed 2023-10-10 by ruslashev