turing-machines/BMC-Firmware #115: Power and status LEDs stop working after upgrading to V1.1.0

mranest, 2023-08-17:
**Describe the bug**
Power and status LEDs stop working after upgrading to BMC firmware V1.1.0. Both work properly with v1.0.2.

**To Reproduce**
Steps to reproduce the behavior:
1. Upgrade to V1.1.0
2. See behaviour of LEDs

**Expected behavior**
After connecting TPi2 to power as soon as BMC finishes booting the status LED should light. The Power LED should light when a node is active.

**Screenshots**
None of the LEDs light up, no matter whether BMC or a node is active.

**Versions**
bmc version=V1.1.0

crkochan, 2023-08-17:
Running `echo "1" > /sys/class/leds/fp:sys/brightness` on the BMC shell illuminates the power button LED connected to the P-LED terminals.

svenrademakers, 2023-08-18:
thanks for reporting! PR is opened here:
https://github.com/turing-machines/tpi_rs/pull/36

and will be part of the next release

closed 2023-09-13 by svenrademakers