turing-machines/BMC-Firmware #118: BMC SD-Card EXT4 don´t works

AndiCorradi, 2023-08-23:
If i try to use an EXT4 formated SD-Card, i got errors on file operations and the Web-Gui->sdcard shows  0 for all values.
i try to copy files with scp (remote host ->  bmc sd-card) but it fails immediateley.

With exFAT, everything works as expected and thr right values are shown in the Web-Gui.

[Documentation](https://docs.turingpi.com/docs/turing-pi2-bmc-sd-card) says that EXT4 is supported.

linux version= 5.4.61
bmc version= 1.1.0

svenrademakers, 2023-09-21:
i prepped an sdcard using the following command:

```
# lsblk
NAME        MAJ:MIN RM   SIZE RO TYPE MOUNTPOINTS
sda           8:0    1     0B  0 disk
sdb           8:16   1     0B  0 disk
sdc           8:32   1     0B  0 disk
sdd           8:48   1  14.9G  0 disk
└─sdd1        8:49   1  14.9G  0 part
# sudo mkfs -t ext4 /dev/sdd1
/dev/sdd1 contains a ext4 file system labelled 'sd_ext'
	last mounted on /mnt/sdcard on Tue Jan  6 08:41:58 1970
Proceed anyway? (y,N) y
Creating filesystem with 3897087 4k blocks and 974848 inodes
Filesystem UUID: cc5a9847-cf9d-44c6-9923-625a604c7ec9
Superblock backups stored on blocks:
	32768, 98304, 163840, 229376, 294912, 819200, 884736, 1605632, 2654208

Allocating group tables: done
Writing inode tables: done
Creating journal (16384 blocks): done

Writing superblocks and filesystem accounting information: done
```

and put it in my board. It seems to load without any issues:

```
# ls /mnt/sdcard/
lost+found
# dmesg
...
[   29.673701] EXT4-fs (mmcblk0p1): mounted filesystem with ordered data mode. Opts: (null)
```

I see the web pages being broken, so we need to fix that for sure.

@AndiCorradi can you tell me more about how you formatted your drive?

AndiCorradi, 2023-09-21:
Hi Sven, i format the sdcard with gparted on ubuntu. I was able to read from the card but not write.

svenrademakers, 2023-09-25:
Could you provide more information?
> i try to copy files with scp (remote host -> bmc sd-card) but it fails immediately

what fails? What is the error message do you see? I have the feeling your SD card is actually not mounted properly. In this case scp will just create the file on the flash, which high likely fails due to the limited space on the flash chip

- what wast the command you used in `parted` to format the drive? did you eject it properly e.g. using `sudo eject /dev/sdX` ? could you insert the sdcard into your computer and post the output of the following command : 
```shell
sudo parted /dev/sdX print
```
you can find the device path by looking it up using `lsblk`

AndiCorradi, 2023-09-25:
I can´t reproduce the error anymore as my board is mounted in a case now with the working exFAT. I guess you are right. Somehow the card did not mount properly. The fail was: i try to copy an image file from my notebook over ssh with scp to the sdcard and it just aborts after a Minute or so. I created the EXT4 partition with the gui version gparted (same as i create the working exFAT). As nobody comment to have the same problem, it is most likely as you said and the issued can be closed.

closed 2023-09-26 by svenrademakers