turing-machines/BMC-Firmware #136: [2.0.0] Power status with RC2 is not clear

meierra, 2023-11-07:
**Describe the bug**
The toggle of power status is not clear if selected means "on" or just selecting to toggle between states.

**To Reproduce**
Steps to reproduce the behavior:
1. Login to BMC url via browser
2. Click on "Power"
3. Scroll down to 'Control the power supply of connected nodes'
4. The on/off buttons have been replaced with select buttons, giving the impression that you are selecting the nodes on which to toggle a change of status, not which status the nodes should change to
[image]

**Expected behavior**
Think the either we should change back to a on/off switch with the submit button executing a change to the state ... or
Have a indicator light to show the current state and have the user select a node to trigger a state change with the submit button.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Versions**
bmc version= 2.0.1

meierra, 2023-11-07:
Additionally a refresh of the page leads to this state, no matter what state the current nodes are in, causing further confusion as to the actual usage of the power page
[image]

svenrademakers, 2023-11-21:
fixed in [v2.0.0-RC3](https://github.com/turing-machines/BMC-Firmware/commit/440838f8a80fd921a64bcb469c62e55fa8448bb1)

closed 2023-11-21 by svenrademakers