turing-machines/BMC-Firmware #137: add support for rockchip USB MSC

svenrademakers, 2023-11-07:
**Is your feature request related to a problem? Please describe.**
The Turing Pi can expose a given node as a mass storage device over USB.  executing `tpi advanced msd --node 1`  will reboot node 1 and load the MSD driver for that given node. Currently, only Raspberry Pi's are supported. The goal of this ticket is to add support for rockchip devices (RK1).

**Describe the solution you'd like**
The `rockusb` driver, created by one of our contributors, is already present in the current firmware and can expose an MSD block device. The work left to do is to update the `bmcd` implementation to use this driver rather than the rockusb rust backend.

**Describe alternatives you've considered**
-

**Additional context**
see https://github.com/turing-machines/bmcd/blob/4c25d6e8a59266f5ee60133a95c1466ffd1e5053/src/firmware_update/rockusb_fwudate.rs#L30

bhuism, 2023-11-07:
Can this potentially work for other rockchip based sbc's? (Like the mixtile) Or is it firmware dependent for instance.

svenrademakers, 2023-11-08:
Potentially yes. The `rockusb` driver talks a proprietary rockusb protocol. It is dependent on the loader implementation of your device how good or not its supported

closed 2023-12-11 by svenrademakers

gjrtimmer, 2023-12-26:
@svenrademakers does the `msd` work with RK1 and also the NVMe ?

gjrtimmer, 2023-12-26:
I saw there where several commits related to have the `msd` work for the `RK1`. Would it be possible to have a new release like 2.0.6 so people who have issues with the `RK1` can use the `msd` to fix there modules.

svenrademakers, 2024-01-04:
Hi @gjrtimmer, we are preparing some other features for the next release. However, the main branch is pretty stable, so it should not be a problem to run a [latest build](https://github.com/turing-machines/BMC-Firmware/actions/runs/7370171126/artifacts/1141026226) on your turing pi.

we load the rockusb driver onto the RK1 in maskrom mode. As far as i can tell it does not support NVME. You would first need to write a bootloader that can interact with a NVME drive to make something like this work.