turing-machines/BMC-Firmware #139: [v2.0.0-RC2] Flashing image for testing / Phoenix Suite crashing

nmaas87, 2023-11-09:
Hello there,
I compiled the image straight from the v2-RC2 tag and got a tp2-bmc-firmware-sdcard.img
Whenever I try to open this file in Phoenix Suite to flash it, Phoenix Suite directly crashes.
Is that expected?
Thanks a lot!

CFSworks, 2023-11-09:
Hello!

Yes, this is expected: the `tp2-bmc-firmware-sdcard.img` file is a *disk* image meant to be written to a microSD card (with a tool like Win32DiskImager, and the card is then used to install the firmware on the BMC), not a PhoenixSuit image. The 2.x series no longer uses any Allwinner-proprietary tools, but I suppose the file extension being the same makes that a bit unclear.

nmaas87, 2023-11-09:
Ah that sounds cool. But how do you trigger the update?
Do you need to have 1.1.0 installed, just flash the img to the SDcard and it will auto flash itself?
Or does a version prior to 1.1.0 suffice?

CFSworks, 2023-11-09:
The microSD card is bootable (think: OS installation USB/CD for PC hardware) and boots the BMC directly to an installation utility when inserted, regardless of the on-board flash contents. This installer will slowly blink the Ethernet port LEDs when ready, and pressing the KEY1 button on the board 3 times in a row will erase the flash and then install the update. The LEDs will strobe rapidly to show that the installation is in progress, which is expected to take about 30-60 seconds. When done, the LEDs will periodically blink twice, and the microSD card should be removed (and the BMC reset) to return to a normal boot.

The first boot will take a few minutes longer than normal while private keys for the SSH and webserver are randomly generated.

nmaas87, 2023-11-09:
Thanks this answers it 💯

closed 2023-11-09 by nmaas87