turing-machines/BMC-Firmware #140: [2.0.0] Kernel Panic After Network Configuration Come Up

bryandph, 2023-11-12:
**Describe the bug**
The BMC is running into a kernel panic shortly after a network interface goes up. This issue showed up as soon as I booted the 2.0.0-RC2 firmware build. On init the BMC was crashing with a kernel panic during SSH and TLS key generation. Notably these both happen right after the network driver initializes and udhcp starts. I unplugged my network interface and was able to get through init and key generation without a crash. 

**To Reproduce**
Steps to reproduce the behavior:
If I boot the BMC with the network adapter attached it eventually crashes with the attached kernel panic.

If I leave the network interface disconnected the BMC boots without issue.

If I reconnect the network interface after init has finished, the BMC eventually crashes, but the nodes and switch remain active.

**Expected behavior**
The BMC should not crash with a kernel panic when the network interface is active. 

**Kernal Panic Dump**
```shell
[123?q]? [  100.820794] skbuff: skb_over_panic: text:8ef2ef30 len:6957 put:6957 head:7715a4f1 data:2eb335be tail:0xc58fa66f end:0xc58f9380 dev:eth0
[  100.834443] ------------[ cut here ]------------
[  100.839585] kernel BUG at net/core/skbuff.c:109!
[  100.844739] Internal error: Oops - BUG: 0 [#1] PREEMPT SMP ARM
[  100.851224] Modules linked in:
[  100.854638] CPU: 0 PID: 0 Comm: swapper/0 Not tainted 5.4.61 #1
[  100.861225] Hardware name: Generic DT based system
[  100.866583] PC is at skb_panic+0x40/0x4c
[  100.870946] LR is at skb_panic+0x40/0x4c
[  100.875323] pc : [<c04eb2d8>]    lr : [<c04eb2d8>]    psr: 60060113
[  100.882289] sp : c0901e28  ip : 00000001  fp : 0000006a
[  100.888120] r10: 00000000  r9 : 00000040  r8 : 00001b2d
[  100.893929] r7 : c58f9380  r6 : c58fa66f  r5 : c58f8b42  r4 : c58f8b00
[  100.901214] r3 : 00000000  r2 : 00000000  r1 : 00000004  r0 : 0000007b
[  100.908471] Flags: nZCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment none
[  100.916439] Control: 10c5387d  Table: 4345806a  DAC: 00000051
[  100.922830] Process swapper/0 (pid: 0, stack limit = 0x678d8007)
[  100.929536] Stack: (0xc0901e28 to 0xc0902000)
[  100.934385] 1e20:                   00001b2d c58f8b00 c58f8b42 c58fa66f c58f9380 c6aa1000
[  100.943522] 1e40: 00000004 c04e4088 c6aa1638 c589ce40 c74476a0 c0419a84 c6aa1738 00000040
[  100.952623] 1e60: c0901e94 c6aa1638 c7ea74c0 00000001 c083e4c0 07669000 c0901e94 0000012c
[  100.961759] 1e80: c0901e9c c04fab48 00000040 ffffb233 c074c42b c0901e94 c0901e94 c0901e9c
[  100.970853] 1ea0: c0901e9c aebc91a5 00000029 c090208c 0000000c 00000008 c0901ed0 c0900000
[  100.979992] 1ec0: c083d780 c08372fc 00000101 c0102ad8 10c5387d c0157fd0 00200002 0000000a
[  100.989100] 1ee0: ffffb232 00000004 40000003 c0902080 c074c42b c083d780 00000000 c7020400
[  100.998238] 1f00: 00000001 c082d238 c0900000 10c5387d 00000000 c011db08 00000000 c01577c8
[  101.007338] 1f20: c0901f48 c8802000 c0904138 c0901f7c c082d238 c0361ff4 c0107fd8 60060013
[  101.016477] 1f40: ffffffff c01021cc 00078820 00000000 00078820 c0114c40 c0900000 c0903de8
[  101.025576] 1f60: 00000001 c0903e24 c082d238 410fc075 10c5387d 00000000 00000066 c0901f98
[  101.034713] 1f80: c0107fe8 c0107fd8 60060013 ffffffff 00000051 00000000 c0900000 c013e684
[  101.043817] 1fa0: 000000cf c0941000 c0903dc0 00000000 c082d238 c013e974 c0941048 c0800ca8
[  101.052951] 1fc0: ffffffff ffffffff 00000000 c0800568 00000000 c082d238 aeb99598 00000000
[  101.062047] 1fe0: c0800330 00000051 10c0387d 00000000 47d33000 00000000 00000000 00000000
[  101.071185] [<c04eb2d8>] (skb_panic) from [<c04e4088>] (skb_put+0x44/0x50)
[  101.078844] [<c04e4088>] (skb_put) from [<c0419a84>] (geth_poll+0x10c/0x190)
[  101.086721] [<c0419a84>] (geth_poll) from [<c04fab48>] (net_rx_action+0x118/0x30c)
[  101.095148] [<c04fab48>] (net_rx_action) from [<c0102ad8>] (__do_softirq+0x1f0/0x278)
[  101.103900] [<c0102ad8>] (__do_softirq) from [<c011db08>] (irq_exit+0x78/0xd0)
[  101.111945] [<c011db08>] (irq_exit) from [<c01577c8>] (__handle_domain_irq+0x74/0xa0)
[  101.120699] [<c01577c8>] (__handle_domain_irq) from [<c0361ff4>] (gic_handle_irq+0x44/0x74)
[  101.129999] [<c0361ff4>] (gic_handle_irq) from [<c01021cc>] (__irq_svc+0x6c/0xa8)
[  101.138353] Exception stack(0xc0901f48 to 0xc0901f90)
[  101.143974] 1f40:                   00078820 00000000 00078820 c0114c40 c0900000 c0903de8
[  101.153109] 1f60: 00000001 c0903e24 c082d238 410fc075 10c5387d 00000000 00000066 c0901f98
[  101.162206] 1f80: c0107fe8 c0107fd8 60060013 ffffffff
[  101.167853] [<c01021cc>] (__irq_svc) from [<c0107fd8>] (arch_cpu_idle+0x1c/0x38)
[  101.176088] [<c0107fd8>] (arch_cpu_idle) from [<c013e684>] (do_idle+0xd0/0x120)
[  101.184255] [<c013e684>] (do_idle) from [<c013e974>] (cpu_startup_entry+0x18/0x1c)
[  101.192682] [<c013e974>] (cpu_startup_entry) from [<c0800ca8>] (start_kernel+0x38c/0x438)
[  101.201820] Code: e58d3014 e5903050 e59f0008 ebf1ad82 (e7f001f2)
[  101.208604] ---[ end trace 447248cb1ec3eff4 ]---
[  101.213758] Kernel panic - not syncing: Fatal exception in interrupt
[  101.220841] CPU1: stopping
[  101.223863] CPU: 1 PID: 0 Comm: swapper/1 Tainted: G      D           5.4.61 #1
[  101.232024] Hardware name: Generic DT based system
[  101.237363] [<c010e390>] (unwind_backtrace) from [<c010a850>] (show_stack+0x10/0x14)
[  101.246023] [<c010a850>] (show_stack) from [<c05e432c>] (dump_stack+0x78/0x94)
[  101.254067] [<c05e432c>] (dump_stack) from [<c010c6d0>] (handle_IPI+0xcc/0x168)
[  101.262237] [<c010c6d0>] (handle_IPI) from [<c036201c>] (gic_handle_irq+0x6c/0x74)
[  101.270666] [<c036201c>] (gic_handle_irq) from [<c01021cc>] (__irq_svc+0x6c/0xa8)
[  101.279025] Exception stack(0xc7097f80 to 0xc7097fc8)
[  101.284651] 7f80: 00042328 00000000 00042328 c0114c40 c7096000 c0903de8 00000002 c0903e24
[  101.293788] 7fa0: 4000406a 410fc075 00000000 00000000 c5fdbf18 c7097fd0 c0107fe8 c0107fd8
[  101.302886] 7fc0: 60070013 ffffffff
[  101.306787] [<c01021cc>] (__irq_svc) from [<c0107fd8>] (arch_cpu_idle+0x1c/0x38)
[  101.315028] [<c0107fd8>] (arch_cpu_idle) from [<c013e684>] (do_idle+0xd0/0x120)
[  101.323196] [<c013e684>] (do_idle) from [<c013e974>] (cpu_startup_entry+0x18/0x1c)
[  101.331624] [<c013e974>] (cpu_startup_entry) from [<40102bec>] (0x40102bec)
[  101.339411] ---[ end Kernel panic - not syncing: Fatal exception in interrupt ]---

```

**Versions**
linux version= Linux version 5.4.61 (runner@fv-az1432-79) (arm-buildroot-linux-gnueabi-gcc.br_real (Buildroot v2.0.0-RC1-3-g3a55f86b) 12.3.0, GNU ld (GNU Binutils) 2.40) #1 SMP PREEMPT Mon Nov 6 17:42:06 UTC 2023
bmc version= Turing Pi 2 BMC Daemon v2.0.1
firmware build= https://github.com/turing-machines/BMC-Firmware/actions/runs/6773906243

svenrademakers, 2023-11-21:
@bryandph Thanks for reporting. My initial thought is that your ethernet device is not entirely up before DHCP kicks in.
Can you share your `/etc/network/interfaces` file?
 Can you say a bit more about your local network? what network services are running? Approximately how many devices are on the network? and do you have multiple turing pi's in your network?
 Did you check obvious MAC or IP collisions?

bryandph, 2023-11-28:
Hi Sven!

Sure; here's the network config (should be the same as buildroot provides):
```shell
#
# cat /etc/network/interfaces
# interface file auto-generated by buildroot

auto lo
iface lo inet loopback

auto eth0
iface eth0 inet dhcp
  pre-up /etc/network/nfs_check
  wait-delay 15
  hostname $(hostname)
```

I have ~ 60 devices on the subnet, and yes, there is a huge variety of multicast traffic hitting the host. The BMC and each compute module have DHCP reservations, so no risk of address exhaustion. If I `watch IP addr` after fully booting up the BMC, and plug-in the network adapter, I can see it get a DHCP lease and can even SSH into it before it eventually crashes with the skbuff kernel panic. Compute modules still stay up and available over the onboard switch, so the issue seems to be limited to the BMC. Only one Turing Pi on the network. I checked for collisions and don't see anything responding to the BMC MAC while the BMC is disconnected or anything concerning in my core switch's ARP table.

Given that the interface actually comes up w/ DHCP and sticks around long enough to SSH into, I think there's a reasonable chance it's reacting badly to something from my LAN. Weird to see a kernel panic if it is the case, and it would be the first device I have seen to react that way.

I can set up a virtual router and a separate LAN to test only the Turing Pi on to confirm the above suspicion. I do have network mirroring capabilities, so if it stays up on the separate LAN I will try to get some PCAPs of what's hitting the BMC while in my lab LAN and see if anything jumps out.

The last build I applied had BMC Daemon v2.0.2, and while I don't expect anything to change, I see you pushed v2.0.5, so I will build that and see if anything changes as well.

bryandph, 2023-11-29:
After rebuilding, I have used the BNC for an extended period without experiencing a crash. Since I can't recreate the issue I will mark as solved and update if anything new occurs.

closed 2023-11-29 by bryandph