turing-machines/BMC-Firmware #141: Add f2fs support to the BMC for more stable SD card operation.

j0ju, 2023-11-14:
**Is your feature request related to a problem? Please describe.**
As a use I like to have a a most stable BMC as it runs idle for the most time, faults are detected when not expected or needed, for wxample when flashing images or using the BMC for out-of-band access for the installed SBC modules.
One source of errors is the SD card used to store images.
The BMC is a long running system, the SD card is used for storing images.
ext[234] filesystems regular write to the same blocks on disks which makes it hard for the embedded flash transaction layer to detect that even if discard or trim are supported by the used SD card. In the end the SD card has errors in case it is most needed.

**Describe the solution you'd like**
Add support f2fs support and tools to the BMC.

j0ju, 2023-12-17:
@gjrtimmer, I do NOT refer to any errors here.
 
I LIKE this feature to support this F2FS included, as it showed in a larger number deployments less errors.
I am also happy to supply patches if there is interest.

@gjrtimmer regarding the issue hijack: yes `noatime` generates fewer writes than `relatime`, `relatime`is the default of the kernel (since 2.6.xx, where xx is the from upper 20ies or lower 30ies)
AFAIK there is no feature in the BMC app the uses the last access time AKA last usage, so it `noatime` is fine to be used.

gjrtimmer, 2023-12-17:
@j0ju sorry my bad, i think it is always great to have community additions. If you can spare the time please submit some patches.

Regarding the noatime. AFAIK its not about if features are used. When any file is accessed then the access time of that file gets updated. But just to be sure I will dive some deeper into it. It has been a while so I want to make sure what the current state of all these options are. Especially since all the SSD advancement in the filesystems and kernel in the last few years.

j0ju, 2023-12-17:
If the BMC app uses the last used time stamps on files or dirs, than `relatime` is needed, if the BMC app does not do this `noatime` is fine. The latter would reduce the amount of writes as for ext[234] FS any meta data update also requires a superblock update, although in ext4 these are delayed (like eventual consistency between super block and dentries).

gjrtimmer, 2023-12-17:
@j0ju Agreed

PR #163 (2023-12-31): configs: add support for f2fs
This adds support for f2fs in the kernel and its tools:

    f2fscrypt
    f2fs_io
    fibmap.f2fs
    mkfs.f2fs
    parse.f2fs
    fsck.f2fs:
        defrag.f2fs
        dump.f2fs
        f2fslabel
        resize.f2fs
        sload.f2fs

Closes #141
commit 602534d5 (2023-12-31): configs: add support for f2fs

This adds support for f2fs in the kernel and its tools:

    f2fscrypt
    f2fs_io
    fibmap.f2fs
    mkfs.f2fs
    parse.f2fs
    fsck.f2fs:
        defrag.f2fs
        dump.f2fs
        f2fslabel
        resize.f2fs
        sload.f2fs

Closes #141
closed 2023-12-31 by svenrademakers