turing-machines/BMC-Firmware #144: persist networking settings during bmc upgrade

fentonfentonfenton, 2023-11-21:
**Is your feature request related to a problem? Please describe.**
When upgrading BMC it would be great if the `/etc/network/interfaces` (etc) files were kept, so that the tpi2 reboots with the same static I.P. etc

I hope it would be fairly easy, if people don't diverge too much from where they set things, and only ever use that file.

---

This is really not a big deal and a nice to have, but if it's a 1 hour ticket or less, then maybe a no brainer.

svenrademakers, 2023-11-21:
Hi,

We are very close to releasing `v2.0.0`. This version can retain network settings and every other modification you make to the filesystem. 
Unfortunately, the upgrade from `v1.x.x` to `v2` will perform formatting of the flash. For this last time, users are required to backup and restore files themselves.

closed 2023-11-21 by svenrademakers

fentonfentonfenton, 2023-11-21:
nice!