turing-machines/BMC-Firmware #146: BMC : *** stack smashing detected ***: terminated

jsilva-iii, 2023-11-23:
**Describe the bug**
Web interface and tpi cli stopped working and are no longer working:

**To Reproduce**
reboot 
**Expected behavior**
Expect to have access to web interface or tpi to power on/off nodes

**Screenshots**
From Serial console:

goahead: 2: Version:            3.6.5
goahead: 2: BuildType:          Debug
goahead: 2: CPU:                arm
goahead: 2: OS:                 linux
goahead: 2: Host:               127.0.0.1
goahead: 2: Directory:          /
goahead: 2: Documents:          /mnt/var/www
goahead: 2: Configure:          me -d -q -platform linux-x86-default -configure . -gen make
goahead: 2: ---------------------------------------------
goahead: 2: Started http://*:80
======= ASP_Init =====func_num 3=====
ASP_Init jst : name app_WebS_Jst_test
 ASP_Init act : name app_WebS_Act_test
 ASP_Init act : name bmc
*** stack smashing detected ***: terminated

tpi cli:
tpi -p off -n 1
curl: (7) Failed to connect to 192.168.0.10 port 80 after 16 ms: Error

**Versions**
linux version= Linux turingpi 5.4.61 #1 SMP PREEMPT Fri Jul 28 18:47:02 UTC 2023 armv7l GNU/Linux
bmc version= 3.6.5
**Additional context**
Was working fine with no issues and after a reboot of the BMC unable to access the web UI

svenrademakers, 2023-11-23:
Duplicate of #75

svenrademakers, 2023-12-20:
this bug is fixed in the [v2 release](https://github.com/turing-machines/BMC-Firmware/releases/tag/v2.0.5)

closed 2023-12-20 by svenrademakers