turing-machines/BMC-Firmware #150: Package Manager

soxrok2212, 2023-11-28:
**Is your feature request related to a problem? Please describe.**
Buildroot uses the opkg package manager. It would be useful to have an official opkg repository with packages that are not necessary, but may be useful for some users (e.g. tcpdump, mtr, etc.).

**Describe the solution you'd like**
opkg support enabled and an official opkg repo stood up!

svenrademakers, 2023-12-06:
I hope the community edition will regain traction soon; it would be a perfect candidate to include.

j0ju, 2023-12-20:
with ansible-openwrt, you can use some basic stuff like file, template, shell and command also without python3. But yes some basic Python3 stuff would be nice for standard BMC, too again.

svenrademakers, 2024-01-04:
i would like to propose we enable it for the SDcard edition:

- [ ] enable the opkg package manager inside buildroot.
- [ ] adjust build pipeline in such a way that we can build both SDcard build and eMMC build with a separate package tree

gjrtimmer, 2025-01-24:
@svenrademakers Does the new 2.1 release has Python again?

svenrademakers, 2025-01-27:
@gjrtimmer I think python would be a good feature for the "community edition" SD card. I dont want to pull it in the official fw as its very big in size. Something we dont have a lot of on turing pi v2.4 boards.

gjrtimmer, 2025-01-28:
@svenrademakers So I'm currently thinking that instead adding Python we add only the package manager, I have already looked into it, and it seems possible that the install location of the additional packages can be on the SDCard. So you can install Python through the package manager and it install it on the SDCard.

What do you think? I'm working on this in PR [Improvements](#232)