turing-machines/BMC-Firmware #160: ADD: Python3

gjrtimmer, 2023-12-12:
**Is your feature request related to a problem? Please describe.**
Python3 is not available therefor using ansible and connecting to the BMC is only possible with raw commands which limits the use of anisble.

**Describe the solution you'd like**
Please add python3 to the buildroot

**Describe alternatives you've considered**
Manual install of python afterwards, not really know how.

soxrok2212, 2023-12-12:
Just my 2 cents. Python 3 is going to be HUGE if it will even fit at all. 

I’ll once again advocate for having a package manager for this: https://github.com/turing-machines/BMC-Firmware/issues/150

gjrtimmer, 2023-12-12:
@soxrok2212 I thought that Python3 was removed, and that it was 19MB.

soxrok2212, 2023-12-12:
Exactly, on a system that is already limited to 128MB, ONE 19MB package for all users (many of which who probably won't use Python directly on it) would be a huge waste.

svenrademakers, 2023-12-13:
Python3 was in one of the previous BMC versions; it slipped in as a dependency for one of the other packages, and the mistake was corrected afterwards.
Its simply too big to have in the firmware. It contributed more than 50% to the overall firmware size!! 

You could enable the package yourself and make a build. But be aware that there will not be enough space to upgrade your BMC via the firmware upgrade component. You would need to use the SD card image produced by your build.

The right place for such packages would be in an alternative firmware edition that runs on the SD card instead of the eMMC. I know there are some plans to kick off such a version.

gjrtimmer, 2023-12-17:
Great to know, the alternative firmware idea sounds very nice. Not be able to upgrade through the firmware component is also undesirable. However having the ability to use ansible with the bmc would really help me. So currently a little torn. Is there a way of installing a packing like python afterwards on the sdcard, or would that only be possible be running the entire firmware from the card. 

So finally the thought that crossed my mind. What about a script like the rk1 has which copies over the os from the eMMC to NVMe, could we use the same here. Like copying most of the bmc out of the eMMC to the SD card and then introduce some kind of package manager?

What are your thoughts, would it be possible regarding the bmc boot process?

svenrademakers, 2024-01-04:
> What about a script like the rk1 has which copies over the os from the eMMC to NVMe, could we use the same here. Like copying most of the bmc out of the eMMC to the SD card and then introduce some kind of package manager?

We can already more or less do this. I created a ticket to install a package manager on the SD-card edition. I think this is low-hanging fruit and would give you the freedom as well to install what you want on your turing-pi. see https://github.com/turing-machines/BMC-Firmware/issues/165

in the meantime, you could try to enable the python package (`echo "BR2_PACKAGE_PYTHON3=y" > tp2bmc/configs/tp2bmc_defconfig`) and copy the target files to a sdcard. You could then point the `PATH` variable to your custom python installation path