turing-machines/BMC-Firmware #164: Consistant naming of Nodes

petermcd, 2024-01-03:
**Describe the bug**
When calling the api/bmc?opt=get&type=power endpoint a response such as the following is returned:

```
{'result': [{'node1': '0', 'node2': '0', 'node3': '0', 'node4': '0'}]}
```

When calling the api/bmc?opt=get&type=usb endpoint the following response is received:

```
{'result': [{'mode': 'Device', 'node': 'Node 3', 'route': 'UsbA'}]}
```

As can be seen, in response to the power call the node is called something like 'node3' however when calling the USB endpoint it is instead called 'Node 3' (namely there is now a space in the node name along with a capital N in node)

This is a minor issue but an unnecessary difference in the naming of nodes.

**To Reproduce**
Steps to reproduce the behaviour:

Calling both API end points will demonstrate the issue.

**Expected behaviour**
Naming of nodes to be consistent

**Screenshots**
n/a

**Versions**
linux version= Linux turing-master-1 5.4.61 #1 SMP PREEMPT Tue Nov 28 13:52:46 UTC 2023 armv7l GNU/Linux
bmc version= 2.0.5

**Additional context**
n/a

barrenechea, 2024-05-14:
My two cents: For an API response, I think it would be better to receive `node1`, `node2`, and so on, as "human-readable texts" should, in an ideal case, be handled by I18n to match the desired user's language.

petermcd, 2024-05-14:
Personally I don't think so. The reason being is there is no unique id for the nodes. In essence the string node1 is the unique id but is inconsistent.