turing-machines/BMC-Firmware #173: LACP Support for network interfaces

haup, 2024-01-16:
** Description**

As a user, i want to be able to utilize the second interface to as part of bond interface for redundancy purposes.

(I couldnt find any regarding open issue, although i was mentioned on one of the announcements in discord for BMC improvments. )

**Describe the solution you'd like**
Prefably configurable on the BMC webUI

hoinzy, 2024-01-18:
This will require configuration of the Realtek Switch, which is mentioned in Issue #46. I would be curious on the progress of this effort aswell.

svenrademakers, 2024-01-21:
it's still on our to-do list. However, we temporarily parked this issue. We are running a somewhat older kernel, which requires a lot of backporting to get the driver in there. Instead, we decided to first upgrade to mainline Linux. Once that's done, we will configure the realtek switch

svenrademakers, 2024-01-23:
i think around a month is realistic

duckman, 2024-07-15:
Any update on this? I know it's not an easy one to solve, but I would love to get some use out of my second port.

rjbrown99, 2025-01-04:
What I'd love to see is the ability to expose 2 different ethernet interfaces to the nodes, one for each physical port on the board. I'd then like to use a crossover cable between two of my Turing Pi 2 boards and run a dedicated 1G network between the two with a large MTU. This would enable a dedicated channel to share network devices across all of the nodes without mixing that with any of the "normal" inbound and outbound traffic. NFS, iSCSI, whatever you want to run as shared storage. 

Seems to be getting closer with the latest firmware. I can now see the br0 bridge interface and how it's configured via /etc/network/interfaces. Just not sure how I'd "un-bridge" the NICs and then expose both to the nodes.

equinox0815, 2025-01-04:
@rjbrown99: I'm not sure about it but i am afraid that the setup that you want to build is not possible with the current state of the switch driver. 

As far as i understand it, this [patch](https://github.com/turing-machines/BMC-Firmware/blob/master/tp2bmc/patches/linux/realtek-switch.patch) basically only introduces 2 new functions to the driver (`port_bridge_join` and `port_bridge_leave`). This allows the kernel to offload bridge membership information to the switch. This means that you can configure a network bridge using the normal tools in linux but what you actually do is to configure the fabric inside the switch chip to directly send packets from on port of the switch to all other member ports.
For example my current network interfaces file looks like this:

```
auto br-mgmt
iface br-mgmt inet static
  bridge-ports ge0
  address 192.168.42.19
  netmask 255.255.255.0
  pre-up /etc/network/nfs_check
  wait-delay 15
  hostname $(hostname)

auto br-svc
iface br-svc inet manual
  bridge-ports node1 node2 node3 node4 ge1
```

This essentially splits up the switch into 2 parts (`br-mgmt` and `br-svc`). The first part contains the first gigabit ethernet port of the mainboard as well as the ethernet port that connects the BMC to the switch. The second gigabit ethernet port is then connected to all nodes. By changing what is listed under `bridge-ports` you can move the corresponding nodes from one network to the other. You might as well introduce i third bridge but since you only have 2 gigabit ethernet ports to the outside world. All nodes which are part of that bridge would only see each other.

However if i understand your use-case correctly you need more than that. Because you want each node to receive and send packets from both sides the nodes need to know which packets belong to which network. With a normal managed switch you would do this using tagged VLAN ports for the nodes. The two gigabit ports on the mainboard would than be configured to each be an untagged member of one of those VLANs. Alas the switch driver has no support for VLAN filtering which means the kernel has no way to offload this VLAN configuration to the switch chip. You can read about this in the [Kernel Docs](https://docs.kernel.org/networking/dsa/dsa.html). The functions that would needed to be implement by the driver are imho: `port_vlan_filtering`, `port_vlan_add` and `port_vlan_del`. But there might be some other stuff needed.
Also the kernel config for the BMC firmware would need be changed to actually turn on VLAN filtering support for bridges (see https://github.com/spreadspace/BMC-Firmware/commit/eca94e4ff81159a7f83a1a8dceeabb17890ea6e0).

My dream setup is actually even more complicated because besides VLAN filtering i would also need bonding to work (`port_lag_join`, `port_lag_leave`). I would than configure the 2 gigabit ports to be bonded together. This bond could than be configured to be a VLAN trunk which in turn would allow me to integrate the Turing Pi board into my existing switch infrastructure and connect any VLAN to any node i want. Either tagged or untagged. 

The good news is that afaik the switch chip should have everything to support such an configuration which means the only thing missing is to extend the driver... I was hoping somebody who has time and the required knowledge to do this would implement this but it hasn't happened yet. This is why i currently use the setup as described above.

rjbrown99, 2025-01-04:
Thanks, helpful input - given the current state of things perhaps what I can do is "un-bridge" both hardware NICs, then bridge node1, node2, and node4 to one NIC, and then bridge node3 to the second NIC. Not exactly what I was hoping for, but if it works it means that the node with the large SSD disk (for an NFS share in my case) has the full 1G bandwidth available to it.