turing-machines/BMC-Firmware #180: Add `tcpdump` (or a dedicated netconsole client) to BMC packages

srcshelton, 2024-01-29:
**Is your feature request related to a problem? Please describe.**

*As a* customer who has a number of Raspberry Pi devices on their network (not all necessarily CM4s in a TP2 board)
*I would like* to make use of the TP2 BMC as my main means to debug all of my RPi devices
*so that* I have all of the diagnostic tools in one place.

**Describe the solution you'd like**

Recent RPi firmware allows network diagnostic logging via the 'NETCONSOLE' configuration option.  It would be great to have the tools to capture and output this on the BMC.

(Exposing this data via the Web UI would be a great additional feature!)

**Describe alternatives you've considered**

Any other linux host _could_ be used… but the BMC seems ideal.

srcshelton, 2024-01-29:
(I note that there's https://web.archive.org/web/20061016094740/http://oss.oracle.com/projects/cancd/, and a more recent but apparently incompatible re-write at https://git.kernel.org/pub/scm/linux/kernel/git/joern/cancd.git, via https://lwn.net/Articles/479674/)

soxrok2212, 2024-01-31:
I can’t imagine how this would be implemented in the web UI, but enabling it on the BMC itself would be helpful for any kind of network troubleshooting. +1 for me.

j0ju, 2024-07-29:
netconsole ist nice, there is already the serial console which can be used for this purposes, too.

gjrtimmer, 2025-01-31:
could the required tools maybe installed when the standard buildroot package manager is included?