turing-machines/BMC-Firmware #181: Rename ttyS* devices to correspond to node numbers

irdc, 2024-01-30:
Currently, the ttyS* devices don't correspond to the node they're connected to, e.g. `/dev/ttyS4` corresponds to node 3. This is not necessary: this is a purpose-built system, and the fact that the BMC's SOC calls a certain UART something doesn't mean that the BMC firmware has to do the same. Renaming them would make things easier on users, who won't have to remember which device corresponds to which node.

I'd like to propose the following mapping:
* `/dev/ttyS0`: for the BMC itself (the connection that's exposed on the Turing Pi as `BMC_UART3`)
* `/dev/ttyS1`: Node 1
* `/dev/ttyS2`: Node 2
* `/dev/ttyS3`: Node 3
* `/dev/ttyS4`: Node 4

MPC-GH, 2024-01-30:
I think this was asked for before and wasn't accepted as a breaking change.

Could I make an alternative suggestion that an alias is put in along the lines of /dev/ttyNode1 though ttyNode4 or similar as an alternative way of sidestepping the ttySx not being Node X issue.

srcshelton, 2024-01-30:
I'd support @irdc's proposal - even accepting that this is a breaking change, if we make it now in the early days of the product then it can be adapted to and we all move on.

The longer the (apparently) random mapping of TTY to node persists, the less possible it will be to fix without the degree of breakage becoming untenable.

I've got a script which maps nodes to ports, and I suspect many others will have followed the same approach - I'd gladly retire this if the port mappings were made human-readable.

Just imagine that there's a hypothetical future Turing Pi board which accepts even more nodes - would it make sense for the non-obvious ordering currently present to persist for all future products (for reasons of consistency) - or would it be best to nip this in the bud for now and the future at this point in time?

CFSworks, 2024-01-30:
See also #8

soxrok2212, 2024-01-31:
I agree with @srcshelton. Best to make the change early.

svenrademakers, 2024-01-31:
It's been a while since #8. A lot of breakage happened going into v2.0. lets update the tty's.

taxilian, 2024-02-16:
Agreed that it can cause some confusion, but I think anyone who is using the TTYs is going to be savvy enough to update things with the change -- and I seriously can't imagine how anyone would not prefer to have them defined in a simple, sane way. I have to look up the mapping every time I use them which is tedious

PR #190 (2024-03-27): Upgrade to Linux 6.6.18
- Unsure about level of compatibility with current `bmcd`
- Completely untested on TPi2.4. All functionality needs to be fully verified before accepting this PR.
  - [x] OTA upgrade works
  - [x] microSD card installation works
  - [x] OTA downgrade to 2.0.x after microSD card installation works
  - [x] CM4 flashing works
  - [x] RK1 flashing works
  - [x] BMC USB port works in device mode
  - [x] BMC USB port works in host mode
  - [x] BMC UART works
  - [x] Keys, LEDs work
  - [x] Serial access to each node works
  - [ ] RTC still works, correct time is set on boot ev [... cut]
commit e7c0a6a5 (2024-03-27): linux: renumber ttyS# devices to a more intuitive order

New mapping:
ttyS0: BMC serial console
ttyS1-4: Serial connection to node 1-4

Closes #181
closed 2024-03-27 by svenrademakers