turing-machines/BMC-Firmware #183: Use a process supervisor

rtreffer, 2024-02-21:
The bmcd service might crash in an OOM situation. It is not restarted as that's not a thing in sys-v-init.

I would like to see/use a proper process supervisor. systemd has one on that front for heavy systems, busybox has stripped down runit implementation.

Shell scripts can be used for that. E.g. mysql famously used a chain of sys-v-init -> mysqld_safe -> mysqld. I would not recommend this, especially if a low disk overhead solution like runit is available.

See e.g. #178 or #162 for a case where the bmcd would stay down.

I have tried to build a BMC with runit, it's quite easy, but probably needs discussion if that's a desirable path forward.
```patch
From bb8a76e86de62a7ca7fe58d5f6eea8d87f4a63d0 Mon Sep 17 00:00:00 2001
From: rtreffer <treffer@measite.de>
Date: Wed, 21 Feb 2024 15:28:49 +0100
Subject: [PATCH] Add busybox/runit to the base image

Runit is a simple process supervisor, suitable for e.g. bmcd.
---
 tp2bmc/configs/busybox.config   | 3 +++
 tp2bmc/configs/tp2bmc_defconfig | 3 +++
 2 files changed, 6 insertions(+)
 create mode 100644 tp2bmc/configs/busybox.config

diff --git a/tp2bmc/configs/busybox.config b/tp2bmc/configs/busybox.config
new file mode 100644
index 00000000..48345897
--- /dev/null
+++ b/tp2bmc/configs/busybox.config
@@ -0,0 +1,3 @@
+# enable runit service supervision
+CONFIG_RUNSV=y
+CONFIG_RUNSVDIR=y
diff --git a/tp2bmc/configs/tp2bmc_defconfig b/tp2bmc/configs/tp2bmc_defconfig
index c50414de..0247398b 100644
--- a/tp2bmc/configs/tp2bmc_defconfig
+++ b/tp2bmc/configs/tp2bmc_defconfig
@@ -60,6 +60,9 @@ BR2_TARGET_UBOOT_FORMAT_CUSTOM=y
 BR2_TARGET_UBOOT_FORMAT_CUSTOM_NAME="u-boot-sunxi-with-spl.bin"
 BR2_TARGET_UBOOT_CUSTOM_DTS_PATH="$(BR2_EXTERNAL_TP2BMC_PATH)/board/tp2bmc/sun8i-t113s-turingmachines-tp2-bmc.dts"
 
+# Busybox customizations
+BR2_PACKAGE_BUSYBOX_CONFIG_FRAGMENT_FILES="$(BR2_EXTERNAL_TP2BMC_PATH)/configs/busybox.config"
+
 # Host packages
 BR2_PACKAGE_HOST_DOSFSTOOLS=y
 BR2_PACKAGE_HOST_GENEXT2FS=y
-- 
2.43.0
```

walkjivefly, 2024-02-21:
God, let's never burden bmcd with anything as unpleasant as systemd.

srcshelton, 2024-02-22:
… why not simply add an entry to `/etc/inittab` with `respawn` declared as the action?

j0ju, 2024-02-28:
if enabling runit tools of busybox also do not forget 'sv' and 'chpst'
The first ist for managing services like starting and stopping, the latter is for writing own .`./run` files for on services.