turing-machines/BMC-Firmware #184: bmcd not reachable if IPv6 only

j0ju, 2024-02-28:
**Describe the bug**
bmcd listens in IPv4 to 0.0.0.0:80 and :443 but on ipv6 is only listens on [::1]:80 and [::1]:443.

**To Reproduce**

Setup an IPv6 only bmc system.
Try to connect to to the bmc's webinterface. --> Connection refused.

'''
tpi: ~ # bmcd -V
2024-02-28T23:22:58.743Z INFO  [bmcd] Turing Pi 2 BMC Daemon v2.0.5
bmcd 2.0.5
tpi:~ # netstat -lntp -6
Active Internet connections (only servers)
Proto Recv-Q Send-Q Local Address           Foreign Address         State       PID/Program name    
tcp6       0      0 :::22                   :::*                    LISTEN      1244/sshd: /usr/sbi 
tcp6       0      0 ::1:443                 :::*                    LISTEN      1249/bmcd           
tcp6       0      0 ::1:80                  :::*                    LISTEN      1249/bmcd           
tcp6       0      0 :::179                  :::*                    LISTEN      1211/bird      

**Expected behavior**
Connection to BMC via V6 is possible.
bmcd is listening on [::]:80 and [::]:443.

**Versions**
linux version= Linux tpi 5.4.61 #4 SMP PREEMPT Sun Jan 28 13:47:02 UTC 2024 armv7l GNU/Linux
bmc version= 2024-02-28T23:49:15.568Z INFO  [bmcd] Turing Pi 2 BMC Daemon v2.0.5

**Additional context**
Might be related to https://github.com/turing-machines/BMC-Firmware/issues/174

svenrademakers, 2024-03-01:
are you running the official 2.0.5 version? if so, https://github.com/turing-machines/bmcd/commit/992c5133474172a0034092642955fe350cdaa4f2 should fix the issue

j0ju, 2024-03-02:
Yeah, just seen it  and bumped to  turing-machines/bmcd@1491cdc to test also the websocket access in my own build. I will report.

svenrademakers, 2024-03-05:
@j0ju just a disclaimer, the websocket implementation is very experimental, and needs some proper testing.

j0ju, 2024-03-07:
I can confirm that the bind/listen behavior now works for ipv4 and ipv6 only hosts :)

closed 2024-03-08 by svenrademakers