turing-machines/BMC-Firmware #185: bmcd logging to console causes strange beahvior if controlling terminal vanished

j0ju, 2024-02-29:
**Describe the bug**
If the controlling TTY of the bmcd process vanishes it behaves strangly on actions.

**To Reproduce**
Steps to reproduce the behavior:
1. Log into the bmc via ssh
2. Stop bmcd via ssh
3. Start bmcd via ssh
4. Log out of the ssh session, ensure it is closes (No Control Master or similar features is in use)
5. Use `tpi advanced msd -n 2`
NOTE: the node number does not matter.

```
tpi: /media/images/rk1 # /usr/bin/bmcd --config /etc/bmcd/config.yaml & pid=$!
2024-02-29T23:26:01.430Z INFO  [bmcd] Turing Pi 2 BMC Daemon v2.0.5

tpi: /media/images/rk1 # tpi advanced -n 2 msd
2024-02-29T23:26:23.406Z INFO  [bmcd::app::bmc_application] Powering off node Node2...
2024-02-29T23:26:24.012Z INFO  [bmcd::app::bmc_application] Prerequisite settings toggled, powering on...
2024-02-29T23:26:25.500Z INFO  [bmcd::app::bmc_application] Checking for presence of a USB device...
2024-02-29T23:26:25.512Z INFO  [bmcd::firmware_update::rockusb_fwudate] Maskrom mode detected. loading usb-plug..
ok

tpi: /media/images/rk1 # echo $pid
24004
tpi: /media/images/rk1 # ls /proc/$pid/fd/[012] -l
lrwx------ 1 root root 64 Feb 29 23:28 /proc/24004/fd/0 -> /dev/pts/3
lrwx------ 1 root root 64 Feb 29 23:28 /proc/24004/fd/1 -> /dev/pts/3
lrwx------ 1 root root 64 Feb 29 23:28 /proc/24004/fd/2 -> /dev/pts/3

[ terminal closed ]
[ fresh terminal ]
tpi: /media/images/rk1 # pid=24004
tpi: /media/images/rk1 # echo $pid
24004
tpi: /media/images/rk1 # ls /proc/$pid/fd/[012] -l
lrwx------ 1 root root 64 Feb 29 23:30 /proc/24004/fd/0 -> '/dev/pts/3 (deleted)'
lrwx------ 1 root root 64 Feb 29 23:30 /proc/24004/fd/1 -> '/dev/pts/3 (deleted)'
lrwx------ 1 root root 64 Feb 29 23:30 /proc/24004/fd/2 -> '/dev/pts/3 (deleted)'

tpi: /media/images/rk1 # tpi advanced -n 2 msd
error sending request for url (https://127.0.0.1/api/bmc?opt=set&type=node_to_msd&node=1): connection closed before message completed
                                                                                                                                                
tipi > ~ > tpi advanced -n 2 normal
ok

tpi: /media/images/rk1 # tpi advanced -n 2 msd
error sending request for url (https://127.0.0.1/api/bmc?opt=set&type=node_to_msd&node=1): connection closed before message completed
```

**Expected behavior**
The operation of  `tpi advanced msd -n 2` should work as expected.

**Versions**
linux version= Linux tpi 5.4.61 #4 SMP PREEMPT Sun Jan 28 13:47:02 UTC 2024 armv7l GNU/Linux
bmc version= 2024-02-28T23:49:15.568Z INFO [bmcd] Turing Pi 2 BMC Daemon v2.0.5

**Additional context**
It seems to be related that the filedescriptors of STDOUT/ERR are closed/vanished when the pseudo terminal of the SSH session vanishes.
If the bmcd is doint output, eg. the state messages regarding the usb devices for msd mode, that it aborts the action without notice.

It would be nice if bmcd would log to syslog. So the messages can the be forwarded to a syslog server, promtail, loki, ...

svenrademakers, 2024-07-29:
having a nice syslog client inside the BMCD would be nice, Off the top of my head, I'm not sure if we have all the pieces in the firmware for this yet. However, the new firmware now logs to a file only by default. 

People who want to have the old behavior back can set the `log.stdout` property to 'true' inside `/etc/bmcd/config.yaml`

@j0ju which terminal client are you using?

j0ju, 2024-07-29:
iterm2, kitty, Konsole, rxvt epending on the workstation I am working. Why you are asking.

Regarding a syslog client: in shell we have the util `logger` which opens and sends messages to /dev/log, which is opened by the syslog deamon. In the current 2.0.5 and later env. this is Busybox syslogd (a companied by klogd)

j0ju, 2024-07-29:
> ... set  log.stdout property to 'true' inside ...
Works as expected.
One wish: if the STDOUT is not a terminal, I like to have no ANSI colour codes emitted.

svenrademakers, 2024-08-02:
There are indeed some libs that can connect to a syslogd. Not much thought is put into designing a good logging system, but I'm planning to start this soon as part of  #153.

however, I've added a config option as well to disable the coloring for now (on default disabled)