turing-machines/BMC-Firmware #186: Power control safety

wiebel, 2024-03-01:
In Version 2.0.5 the way to power on and off the nodes is WAY to direct.

A better solution would be 
 - checkboxes for the nodes
 -  seperate button for "Power on" "Power off" and maybe even "Powercycle"
 - an "Are you sure?" Dialog

wiebel, 2024-03-01:
It's probably more relates to https://github.com/turing-machines/BMC-UI
Or more precisely: https://github.com/turing-machines/BMC-UI/blob/main/src/assets/js/tabs/tab-nodes.js
But as it seems to be packaged within this firmware release it seems more relevant here.

svenrademakers, 2024-03-23:
Hi @wiebel, you submitted the ticket to the correct repository. 

related to #169

svenrademakers, 2024-07-22:
fixed in e5f6158eeb61449a6b751e8febd27c6069cd56bb and onwards

closed 2024-07-22 by svenrademakers