turing-machines/BMC-Firmware #187: Issues flashing CM4 slot 1?

deosrc, 2024-03-18:
**Describe the bug**

When trying to flash a Raspberry Pi CM4 in slot 1 using rpiboot, something seems to be preventing the USB OTG connection.

**To Reproduce**

1. Connect a USB A-A cable between a machine and the Turing Pi 2.
2. With a CM4 with eMMC memory in slot 1, go to the UI.
3. Power down node 1.
4. Change the USB mode to "Flash" on node 1.
5. Run `rpiboot -d ./msd`
6. Power up node 1.

Additionally attempted:

1. Connect a USB A-A cable between a machine and the Turing Pi 2.
2. Login to the Turing Pi using ssh.
3. Power down node 1: `tpi power -n 1 off`
4. Change the flash mode: `tpi usb -n 1 flash`
5. Confirm using `tpi usb status` (Outputs `usba --> node 1`)
5. Run `rpiboot -d ./msd` on connected machine.
6. Power up node 1: `tpi power -n 1 on`

Have also attempted running `tpi advanced -n 1 msd` at various stages.

Haven't yet attempted a different node slot. I will attempt and report back to ensure it is not a cable or connected machine issue.

**Expected behavior**

- lsusb should show an additional device connected
- rpiboot should detect and push the boot code to the CM4 in node 1

**Versions**
linux version= Ubuntu 22.04.4
bmc version= 2.0.5

**Additional context**

Related discussions:

- https://docs.turingpi.com/discuss/65eab8a6aa22e000363bb546
- https://docs.turingpi.com/discuss/65859094d3e7aa0038342c74

gjrtimmer, 2025-01-28:
@deosrc, an important question: Is the attempt to flash using a bare metal machine or virtual machine?

Furthermore, I personally can attest to the fact that the cable might be an issue. While some USB-A cables only provide power and not data transfer, and while these cables are rare, it is still possible. I had such a cable.

I also had an issue with flashing in node 1 until I used a bare metal machine, and suddenly, the issue no longer existed. Tried Virtual Machine and macOS first.

Would you happen to have any more information yet?

gjrtimmer, 2025-01-28:
Marking issue as stale

deosrc, 2025-01-28:
It was a bare metal machine. I can't remember what happened with this but I guess I managed it somehow as I have had 3 nodes in the turing pi as part of my k3s cluster for a while now.

It may have been a port issue on the flashing machine. I have a Framework 13 AMD laptop and have had some strange behaviour from the expansion cards. Some peripherals have been picky about which ports they will work with. There are documented [expansion card limitations on some of the ports](https://knowledgebase.frame.work/en_us/expansion-card-functionality-on-framework-laptop-13-amd-ryzen-7040-series-SkrVx7gAh). It doesn't seem to mention OTG but I'm not very familiar with the protocols involved.

closed 2025-01-28 by deosrc