turing-machines/BMC-Firmware #196: Configure Network Switch from BMC

adesutherland, 2024-04-16:
I am adding this request because there does not seem to be any issues relating to configuring the on-board switch from BMC, please feel free to remove this if I have missed something.

The request it to as a minimum set VLAN restrictions on a per port basis, and perhaps other switch management features. 

I understand that this may be blocked by waiting for an upstream Linux version, or something like that. Anyway hopefully this issue will allow this feature to be tracked :-)

flokli, 2024-04-17:
https://github.com/turing-machines/BMC-Firmware/issues/173 is at least related. Both VLAN and LACP (even in combination) are things the switch chip can be configured for.

ludwigkraatz, 2024-11-30:
fyi: This is the one thing keeping me from investing into a turing pi board..

any updates on a roadmap regarding that feature?

equinox0815, 2025-01-28:
@gjrtimmer i documented [here](https://github.com/turing-machines/BMC-Firmware/issues/173#issuecomment-2571408718) what i think is missing for the switch to be fully configurable. Long story short: it's mostly the state of the kernel driver that prevents this from happening. Changes in `bmcd` will later be necessary to make all this configurable through the web interface but a working kernel driver that implements all the necessary callbacks is required first.

j0ju, 2025-06-16:
I investigated some work into split configuartion of all ports on the bmc:
tipi > / > ls -l /etc/network/interfaces /etc/network/interfaces.d/* ; for i in /etc/network/interfaces /etc/network/interfaces.d/* ; do echo '```'; echo "# $i"; cat $i; echo '```'; done
-rw-r--r-- 1 root root 850 Jun 14 01:35 /etc/network/interfaces
-rw-r--r-- 1 root root 242 Jun 14 01:35 /etc/network/interfaces.d/eth0.DSA
-rw-r--r-- 1 root root 187 Jun 14 22:11 /etc/network/interfaces.d/ge0
-rw-r--r-- 1 root root 245 Jun 14 01:35 /etc/network/interfaces.d/ge1
-rw-r--r-- 1 root root 335 Jun 16 11:35 /etc/network/interfaces.d/mgmt
-rw-r--r-- 1 root root 244 Jun 14 01:35 /etc/network/interfaces.d/node1
-rw-r--r-- 1 root root 244 Jun 14 01:35 /etc/network/interfaces.d/node2
-rw-r--r-- 1 root root 244 Jun 14 01:35 /etc/network/interfaces.d/node3
-rw-r--r-- 1 root root 215 Jun 14 22:29 /etc/network/interfaces.d/node4
-rw-r--r-- 1 root root 843 Jun 14 01:35 /etc/network/interfaces.d/sw0
```
# /etc/network/interfaces
# /etc/network/interfaces
#
# Basic network config for turinppi2 using ifupdown-ng
# The config is split into serveral snippets below
# /etc/network/interfaces.d/
#
# * for enabling DSA (distributed switching architecture) to have VLAN and
#   Bonding/LACP offloading see
#     * sw0 - here you can change the MGMT VLAN config
#       * MGMT VLAN is currently unttaged on ge0/ge1 and mapped to VID 4094
#         at the end of the VID range
#     * eth0.DSA - only modify if you know what you are doing

# * for changing mgmt port address and VID see
#   * mgmt
#
# * for changing trunkport VLANs see
#   * ge0
#   * ge1
# * for changing nodeport VLANs see
#   * node[1..4]

# loopback interface do not change
auto lo
iface lo inet loopback

# include dynamic config
source-directory /etc/network/interfaces.d

# vim: ts=2 sw=2 et foldmethod=indent
```
```
# /etc/network/interfaces.d/eth0.DSA
# control interface for DSA switch ASIC
# do not add l3 addresses, it needs to be up, for switch offloading to work
auto eth0
iface eth0
  # no IP addresses on this interface
    force-no-l3-address yes

# vim: ts=2 sw=2 et foldmethod=indent
```
```
# /etc/network/interfaces.d/ge0
iface ge0
  # allow untagged frames for eg mgmt
    bridge-allow-untagged yes
    bridge-pvid 4094

  # specify tagged VLANs on port
  # VID
  #   6 - FFHO via backyard
    bridge-vids 6
```
```
# /etc/network/interfaces.d/ge1
iface ge1
  # allow untagged frames for eg mgmt
    bridge-allow-untagged yes
    bridge-pvid 4094
  # specify tagged VLANs on port
  # NOTE: this line need to be adpoted by bmcd for VLAN config
    bridge-vids # 2 3 5 7 11 13 - currently empty
```
```
# /etc/network/interfaces.d/mgmt
# /etc/network/interfaces

# managment access to BMC

auto mgmt
iface mgmt
    use ipv6-ra
  # set VLAN parameters
    vlan-raw-device sw0
    vlan-id 4094
  # address config
    ipv6-ll no
    ipv6-autoconf no
    address fe80::bc/64
    address fd80:141:314::bc/64
    dns-nameservers fd53::53

# vim: ts=2 sw=2 et foldmethod=indent
```
```
# /etc/network/interfaces.d/node1
iface node1
  # untagged 4094
  # NOTE: this needs adoption by bmc to configure an untagged VLAN
  bridge-pvid 4094
  # tagged VLANS
  # NOTE: this needs adoption by bmc to configure tagged VLANs
  bridge-vids # 2 3 5 8 13 21 - currently empty
```
```
# /etc/network/interfaces.d/node2
iface node2
  # untagged 4094
  # NOTE: this needs adoption by bmc to configure an untagged VLAN
  bridge-pvid 4094
  # tagged VLANS
  # NOTE: this needs adoption by bmc to configure tagged VLANs
  bridge-vids # 2 3 5 8 13 21 - currently empty
```
```
# /etc/network/interfaces.d/node3
iface node3
  # untagged 4094
  # NOTE: this needs adoption by bmc to configure an untagged VLAN
  bridge-pvid 4094
  # tagged VLANS
  # NOTE: this needs adoption by bmc to configure tagged VLANs
  bridge-vids # 2 3 5 8 13 21 - currently empty
```
```
# /etc/network/interfaces.d/node4
iface node4
  # untagged 4094
  # NOTE: this needs adoption by bmc to configure an untagged VLAN
  bridge-pvid 4094

  # tagged VLANS
  # NOTE: this needs adoption by bmc to configure tagged VLANs
  bridge-vids # 6
```
```
# /etc/network/interfaces.d/sw0
# switch / bridge interface
auto sw0
iface sw0
  # MUST not add eth0 here
    bridge-ports node1 node2 node3 node4 ge0 ge1
  # enable VLAN awareness
    bridge-vlan-aware yes
  # no IP addresses on this interface
    force-no-l3-address yes
  # configure default VLAN
  #  * egress traffic is untagged
  #  * ingress untagged traffic is put to into VLAN1
    bridge-allow-untagged no
    bridge-default-pvid no
  # untagged vlans - the mgmt VID needs to be added
  # mgmt VID is 4094 at the end of the VLAN id range, to vaoid collision with
  # other VLANs tpycally count from lowest 1
    bridge-vids 4094
  # enable spanning tree if needed
    bridge-stp off

  # allow untagged frames
    bridge-allow-untagged no
  # expose managemnt VLAN tagged so we can create mgmt interface
    bridge-vids 4094

# vim: ts=2 sw=2 et foldmethod=indent
```

equinox0815, 2025-06-16:
This is very nice setup. For my dream setup i would like to also be able to join `ge0` and `ge0` into an LACP bond and then use this bond as an uplink for all VLANs. But besides that i think the configuration layout looks promising.
Now we only have to wait for somebody to implement the necessary callbacks in the switch driver to offload the VLAN tagging to the switch chip to make it vlan-aware.

j0ju, 2025-06-16:
For bond I have an idea, which is working with an active/passive setup, but I could net test link bandwitth aggregation (2x 1G = 2g) as I have no fiting HW for the counter part at home.
I introduce a bond0 interface similar to aboves split config which is then added to the bridge.
If the ge0/1 are moved to the bond by modifying the files for bond0 and sw0 by the BMC (or for now manually) it works.
The neccesary hooks in the call back driver are already present for VLAN for sure, and it is offloaded. For bonding/LACP I am unsure.

j0ju, 2025-06-16:
with this split of config the UI and API of the BMC can just modify single files and restart the network.
valid attibutes to change ....
* on VLAN config change on ge0/1 bridge-vids and bridge-pvid of the related files
* on VLAN config change on node1234 bridge-vids and bridge-pvid
* if the MGMT vlan VID 4094 needs  to be changed all files
* ip change just mgmt

Bonding is a special case I am currently puzzling about.
Currently it introduces a bond0 config for an interface of the same name.
For enable bonding it would include the bond0 into sw0 and removing ge0/1 and adding it as member interfaces to bond0.

equinox0815, 2025-06-17:
> For bond I have an idea, which is working with an active/passive setup, but I could net test link bandwitth aggregation (2x 1G = 2g) as I have no fiting HW for the counter part at home. I introduce a bond0 interface similar to aboves split config which is then added to the bridge. If the ge0/1 are moved to the bond by modifying the files for bond0 and sw0 by the BMC (or for now manually) it works. The neccesary hooks in the call back driver are already present for VLAN for sure, and it is offloaded. For bonding/LACP I am unsure.
> 
> I tested VLAN offloading by adding multiple indiviual VLANs tagged and as pvid/untagges to a node. I could download via the untagged and tagged vlans with 1Gbit/s. This is more as the BMCs 100Mbit/s speed. I the switchging wiht tagged VLANs would happend in the t113-s3 I would expect only 100Mbit/s.

hmmmm - this is very interesting! I thought the switch driver must implement the callbacks `port_vlan_filtering`, `port_vlan_add` and `port_vlan_del` in the `dsa_switch_ops` struct in order to make it vlan-aware. Apparently this is not the case. I've been starring at the driver code for an hour now and still have no clue how this works. The best theory i came up with is what is hinted at [here](https://www.kernel.org/doc/html/v6.8/networking/dsa/dsa.html#switchdev) and [here](https://www.kernel.org/doc/html/latest/networking/switchdev.html#l2-forwarding-offload). Maybe in case the for-mentioned callbacks are not implemented DSA will fallback to just offload the FDB entries to the switch... somehow. But this IMHO means that the Ethernet frames from an unknown MAC address will initially be forwarded to the CPU and once the kernel has added the MAC address to its forwarding database it will be synced with the switch which than handles the subsequent frames directly. Not sure if this true, but if this is the case i don't like it. This would mean that if the kernel of the BMC hangs for some reason the communication to the nodes might fail in a very subtle, hard to debug, way.
However in the coming days i should find a little bit of time to try out your config setup on my TuringPi - including some tests with bonding and LACP.

equinox0815, 2025-06-22:
I did some tests with your setup today but unfortunately had no success in reproducing your results. For my tests i first copied your configuration files (almost) verbatim to `/etc/network/` and tried a reboot. After the reboot i could not reach the Turing PI anymore. Using the USB-debug serial port i tried some variations of vlans and commenting out some stuff but alas it didn't work. So i started debugging the configuration piece by piece. I saw that the bridge interface wasn't showing up in the interface list so i tried to start it manually using `ifup sw0` but nothing happened. So i tried to create it myself using `ip link add sw0 type bridge` followed by `ip link set dev sw0 type bridge vlan_filtering 1`. The last command didn't work: `RTNETLINK answers: Operation not supported`.

It was then that i remembered that the current default kernel config of the Turingpi BMC firmware has bridge vlan-filtering disabled. Some time ago i built a custom kernel to circumvent this issue (https://github.com/spreadspace/BMC-Firmware/commit/eca94e4ff81159a7f83a1a8dceeabb17890ea6e0). Alas this kernel was based on an older firmware image that used a different kernel version. So i would need to build a new kernel based on the current version that is used by the firmware and i didn't have enough time this weekend to do this.

But this begs the question: how did you test your setup? Have you used your config only inside of your docker based environment? If yes i don't believe that this is a valid approach to fully test this configuration. All of the bridge VLAN offloading stuff is highly depended on what features are built into the kernel as well as the hardware it is running on. But didn't you also mention that you had speed tests running to verify that the traffic doesn't go through the 100Mbit/s interface of the BMC? Which firmware/kernel were you running for those tests?

equinox0815, 2025-12-09:
> Of course you should see braodcast and multicast traffic on ports where the VLANs are configured on. In this case I see VLAN traffic for those packets on all ports.

hm yeah this shouldn't be the case.

j0ju, 2025-12-11:
tl:dr: VLANs 802.1q will not work fully until the driver is extended.

@equinox0815  Okay after I looked deeper into the issue I agree with your summary  from https://github.com/turing-machines/BMC-Firmware/issues/196#issuecomment-2994275583.
The current driver from 6.8 cannot do VLAN handling. The needed callback methods are missing, funny thing you can set "vlan_filtering", but this assumes that the switch ASIC is sending all traffic to the CPU port and this is done there.
If I interpret the code correctly, the driver is configuring it as a switch, can even to two bridges like
sw0 asigned to ge0,node1 and node2 and 
sw1 assigned to ge1, node3/4 and you have isolation. But this just port isolation without VLAN groups properly configured.
In the driver even code that gives hints that dot1q is not support, as they use VLAN config for port isolation, but without any proper VLAN tagging etc.
The driver is nearly the same for current 6.18 vanilla, 6.8 from TP2 Buildroot and even in OpenWRT git-HEAD.
@CFSworks did a great job in doing the needed adaption to get the driver working SMI/I2C on the BMC bord.

The tests and measurements I just relied on the switch behaviour with MAC lookup tables for egress ports, and were biased and not conclusive.

Some comments from the driver `bmc-firmware/buildroot/output/build/linux-6.8.12/drivers/net/dsa/realtek/rtl8365mb.c`
``` 
...
static int rtl8365mb_port_set_trapping(struct realtek_priv *priv, int port,  bool enable)
{
  int ret;

  /* Prevent ingress traffic from being handled by the switch fabric by
   * using dot1x mode. The dot1x functionality isn't used elsewhere in
   * the driver, so is a convenient method for trapping ingress traffic.
   */
...

```