turing-machines/BMC-Firmware #200: The node UART functionality of the BMC-Firmware-master branch (internally v2.1.0-unstable-2) is not working

dgdonovan, 2024-05-06:
**Describe the bug**
Node UARTs do not provide output. The BMC UART and picocom from the BMC to each node does work.

**To Reproduce**
Steps to reproduce the behavior:
1. Confirm supported SoMs are plugged into the TPiv2 and have a functioning OS installed.
2. Confirm that all nodes are powered off.
3. Attach the USB-A male end of a USB to UART/TTL cable to a Windows 10 system with PuTTY installed.
4. Confirm through Windows devmgmt.msc tool that the USB to UART/TTL cable is recognized and registered as a COM port (note the COM port)
5. Configure a PuTTY session for the appropriate Serial Line (COMx) with a Speed of 115200, then click on the "Open" button.
6. Attach the other end of the USB to UART/TTL cable to the appropriate node [2,3,4] UART pins on the TPiv2.
7. Open two (2) PuTTY SSH sessions to the BMC's IP address.
8. From one SSH session, enter `picocom /dev/ttyS[2,3,4]`.
7. Use `tpi power --user root --password turing --node [2,3,4] on` from the second BMC SSH session to power the appropriate node on.
8. The BMC SSH running picocom should display boot output
9. The PuTTY connection to the node's UART port does not display anything.

**Expected behavior**
A node's UART port, which is intended for debugging, should display all boot output and present the user with a logon prompt.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Versions**
linux version= <output of `uname -a`>
bmc version= <printed at startup>

**Additional context**
Add any other context about the problem here.

svenrademakers, 2024-07-29:
hi @dgdonovan, which modules did you use? im not sure if the uart ports to the second tty device are enabled by default on for instance a raspberry pi

dgdonovan, 2024-07-29:
@svenrademakers I believe this was either with an RK1 or an NVIDIA Jetson Orin NX 16GB module.

In testing a Radxa CM5 (RK3588S2 in CM4 form factor), I found mention in a Radxa documentation page over the weekend that RK35XX UARTs run at 1500000, not 115200, and that UART-to-USB cables based on CP210X, PL2303x (I tested with one of these) chips have a baud rate limitation and cables based on FT232RL chip have power issues.

I ordered and received cables based on the recommended CH340 chip yesterday and will check one with an RK1. If this fixes the issue, I'll close this issue and check the RK1 documentation to see whether it includes complete instructions.