turing-machines/BMC-Firmware #201: [v2.5] batch flash nodes

svenrademakers, 2024-05-13:
**Is your feature request related to a problem? Please describe.**
The Turing Pi can flash an OS image to a given (supported) module. The firmware loads a USB plug onto the module, which, in turn, exposes an API used to write the new OS images. On v2.4 boards, only one module can be switched to the BMC at a time. On v2.5, we replaced these muxes with a USB hub, which opens up the possibility of flashing multiple nodes simultaneously.

**Describe the solution you'd like**
We want to be able to write an image to a selection of nodes.  The dropdown in the "flash" tab of UI gets replaced with checkboxes, which the user can use to select which nodes to flash simultaneously. Flashing different images concurrently to different nodes is out of scope. Keep it simple!
* if an error occurs with one of the nodes, all other tasks are aborted as well.
* error messages need to be altered so it's clear to the user which node caused the error.
* All flashing features should be expanded to the other nodes as well. sha265 checking, skip crc bool and xz decompressing. Be mindful that we are extremely limited on memory. for instance, don't decompress the same OS image multiple times. 

**additional information**
we expect changes in the following 2 repos:
- [BMC-UI](https://github.com/turing-machines/BMC-UI)
- [BMCD](https://github.com/turning-machines/bmcd)

barrenechea, 2024-05-13:
I can prepare the UI so we're able to support this use case, I'll be playing with options there 😄

barrenechea, 2024-05-13:
To know, the UI should behave differently depending on if the board is < v2.4 (or >v2.5). Am I right? Is there a chance that I could get the board revision to render options conditionally? I think a good endpoint would be the one currently providing data for the `About` tab

svenrademakers, 2024-05-14:
You brought up a good point. Of course, this behavior should only occur when a 2.5+ board is detected.  You're also right that we need an endpoint to detect which of the 2 versions needs to be loaded. I would prefer to have a field encoded in the actual flashing endpoint that specifies something like: 
```
{
 can_do_bulk_flashing: true
 ...
 }
```
Making the code dependent on the firmware version is a less clean option as we make ourselves dependent on this specific hardware when in theory, it doesn't matter on which hardware it runs.

> I think that a good option would be for v2.4 users only to be able to pick one option (and automatically disable the user from picking more than one choice), 

that sounds good to me. it will keep things consistent!

barrenechea, 2024-05-17:
@MPC-GH Yeah you're right, it probably streams the uploaded file directly to the target node(s). Better to keep it simple for now so we don't delay the main feature.

@svenrademakers a question regarding the `/api/bmc?opt=set&type=flash` call. Currently, it expects something like:
`/api/bmc?opt=set&type=flash&file=ubuntu.img&length=55345150&node=0` (node being 0-indexed)

Would it make sense for this to send a comma-separated list in the `node` field for bulk flashing? Something like:
`/api/bmc?opt=set&type=flash&file=ubuntu.img&length=55345150&node=0,1,2,3`

svenrademakers, 2024-05-23:
@barrenechea, I would like to keep the API backward compatible as much as possible. Therefore, it would be better if we introduced an additional key (it's not more elegant by all means). Maybe `copy` or `batch` is the right word?

e.g.
`/api/bmc?opt=set&type=flash&file=ubuntu.img&length=55345150&node=0&batch=1,2,3`