turing-machines/BMC-Firmware #206: `tpi flash ...`: `Failed to Compute module not supported: Compute module not supported` Search  `tpi flash ...`: `Failed to Compute module not supported: Compute module not supported` chat `tpi flash ...`: `Failed to Compute module not supported: Compute module not supported`

j0ju, 2024-07-08:
**Describe the bug**
I am testing currently a firmware and bmcd build from git master(dc91804d)  and I own an TPi2 from the Crowdfound process.

This issue is linked to https://discord.com/channels/754950670175436841/1259272804017246369/1259272804017246369
as @CFSworks suggested it might be a regression.

I am getting above error message while turning any node rk1 or cm4 in to usb storage mode via 
```
# tpi flash -l -i $PWD/lite-mdns-networkd.64.img -n 1
Flashing from image file /mmc/CM4/lite-mdns-networkd.64.img...
Error occured during flashing: "Compute module not supported"

[... bmcd log ...]
2024-07-06T22:10:53.097763Z  INFO bmcd::streaming_data_service: #3452941224 'Node 1 os install service' 606.66 MB - started
2024-07-06T22:10:53.098522Z  INFO bmcd::app::bmc_application: Powering off node Node1...
2024-07-06T22:10:53.201704Z  INFO bmcd::app::bmc_application: changing usb config to Flashing(Node1, Bmc)
2024-07-06T22:10:53.201902Z  INFO bmcd::app::bmc_application: Powering on...
2024-07-06T22:10:54.690905Z  INFO bmcd::usb_boot: Checking for presence of a USB device...
2024-07-06T22:10:54.692790Z ERROR bmcd::streaming_data_service: #3452941224 stopped: Compute module not supported.
```
the same happens with advanced ... msd and is independent of which node was used.
It does not matter if it is a CM4 module or an RK1.
```
# tpi advanced -n 1 msd
 INFO bmcd::app::bmc_application: Powering off node Node1...
 INFO bmcd::app::bmc_application: changing usb config to Flashing(Node1, Bmc)
 INFO bmcd::app::bmc_application: Powering on...
 INFO bmcd::usb_boot: Checking for presence of a USB device...
Failed to Compute module not supported: Compute module not supported
```
In both cases the target node turns on but tpi uart get -n 1 does not show any output ￼ 
# uname -a
Linux tipi 6.6.18 #3 SMP PREEMPT Fri Jul  5 17:13:04 UTC 2024 armv7l GNU/Linux

Generally this worked before with a version I compiled and installed during end January 2024.

Maybe this is related? Additionally i found that this data is not correct in newer tpi/bmcd versions, maybe a hardware usage/dtb mismatch, as I have 2.4.0 from the crowdfounding
The uboot is booting with the dtb for 2.4.0.

But bmcd thinks it is a 2.5.2
```
# tpi eeprom get
BoardInfo {
    reserved: "0xffff",
    crc32: "0xe66e5ba5",
    header version: 65535,
    hardware version: "v2.5.2",
    factory date: 2023-05-01,
    factory serial: "����������������",
    product name: "����������������",
    mac: "ffffffffffff",
}
```
The hypothesis would be that bmcd cannot to the USB muxing/switching properly as 2.5.x and 2.4.0 differ in that area.

svenrademakers, 2024-07-09:
@j0ju
to determine if linux is running the correct version, print the model attribute of the DT:
```shell
cat /proc/device-tree/model
```
(2.5.1 == 2.5.2)

regarding the EEPROM contents, you probably tried out the factory image or ran 'burn_eeprom' in the bootloader at some point. Which by default is set to 2.5.2.
It's pure happenstance that your bootloader loads 2.4, as I changed the EEPROM format around that time, which probably means that the bootloader generates a mismatched crc, causing it to fall back on loading 2.4.
I would recommend updating your EEPROM with the correct version (2.4). run the following in uboot:

```shell
env set eeprom_ver 0x1100; run burn_eeprom
```

I've heard of hardware failures on the Turing-pi board, causing the RK1 to be undetectable by the firmware @daniel-kukiela?

When you get served with `failed to Compute module not supported: Compute module not supported`, it means that the USB device (with vid/pid 2207:0x350 for RK1) it is not showing up in Linux. This is the entry point for the firmware to start loading the USB plug..

j0ju, 2024-07-09:
Thanks, this explains it a bit, although I have not run the factory image. (but I looked at the code, though)

Currently
```
# cat /proc/device-tree/model
Turing Pi 2 (v2.5.1)#
```

After booting into uboot and executing the suggested command 
```
env set eeprom_ver 0x1100; run burn_eeprom
```
I get 
```
# cat /proc/device-tree/model
Turing Pi 2 (v2.4)#
```
So this might indicate that somehow my eeprom contents were messed with.

This looks better now:
```
# tpi eeprom get
BoardInfo {
    reserved: "0xffff",
    crc32: "0x33b5f180",
    header version: 65535,
    hardware version: "v2.4.0",
    factory date: 2023-05-01,
    factory serial: "����������������",
    product name: "����������������",
    mac: "ffffffffffff",
}
# tpi advanced msd -n 1
 INFO bmcd::app::bmc_application: Powering off node Node1...
 INFO bmcd::app::bmc_application: changing usb config to Flashing(Node1, Bmc)
 INFO bmcd::app::bmc_application: Powering on...
 INFO bmcd::usb_boot: Checking for presence of a USB device...
 INFO bmcd::usb_boot: ID 0x0a5c:0x2711 rustpiboot
 INFO bmcd::usb_boot::rpiboot: Checking for presence of a device file ('RPi-MSD-.*')...
 INFO bmcd::utils: Stopping bmc-otg: OK
 INFO bmcd::utils: Starting bmc-otg: OK
 INFO bmcd::app::bmc_application: BMC-OTG: Node mass storage CDC enabled
ok
```

closed 2024-07-09 by j0ju