turing-machines/BMC-Firmware #217: thread 'main' panicked at src/persistency/binary_persistency.rs:

meierra, 2024-08-19:
**Describe the bug**
After upgrading to RC2 for 2.5 firmware, the BMCD fails to start and turingpi is not accessible from https 443, no errors in /var/log/messages that seem to be related to the issue.

Manually starting the bmcd (ssh terminal works)
bmcd --config /etc/bmcd/config.yaml

```
thread 'main' panicked at src/persistency/binary_persistency.rs:175:33:
called `Result::unwrap()` on an `Err` value: SerializationError("node_info", InvalidTagEncoding(4))
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

**To Reproduce**
Steps to reproduce the behavior:
1. Upgrade to RC2 of 2.5
2. Reboot
3. login to ssh
4. bmcd --config /etc/bmcd/config.yaml
5. See error

**Expected behavior**
bmcd starts

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Versions**
linux version= Linux turingpi 6.8.12 #1 SMP PREEMPT Thu Aug 15 07:54:25 UTC 2024 armv7l GNU/Linux
bmc version= bmcd 2.3.0

svenrademakers, 2024-08-19:
It seems there is a binary incompatibility in the storage file.
from which version did you upgrade from?
could you post the storage file. its located on '/var/lib/bmcd/bmcd.bin'
also a copy of the logging file would be helpful, its in '/tmp/bmcd.xxxx-xx-xx.log'

As a work-around:
1. login to the bmc
2. remove file '/var/lib/bmcd/bmcd.bin'
3. run '/etc/init.d/S94bmcd restart'

meierra, 2024-08-19:
Thanks that worked, I updated from 2.0.5, I have uploaded the bin file, does it get recreated?
[bmcd.zip](https://github.com/user-attachments/files/16663395/bmcd.zip)

svenrademakers, 2024-08-20:
yes it will create a default one

commit b4bb7bc4 (2024-08-20): bmc-ui & bmcd: increase patch ver

fixes #217
closed 2024-08-20 by svenrademakers