turing-machines/BMC-Firmware #220: Beta testing firmware updates

rjbrown99, 2024-09-03:
We're coming up on a year since the last firmware release. How are you thinking about a process for the adventurous souls out there to test more recent BMC code changes? IE would you consider either directly publishing an "alpha" or "beta" version with details on how to test and report bugs, including what board revision they are being tested on? 

And if not, would you at least let us know which specific date/commit you might be considering as a code freeze for a release so we could compile our own to test?

There are a few things introduced in the past 10 months I'd like to try out. Fan control is a big one. I understand how to compile firmware, but what I don't currently understand is if any changes break the ability to roll back to 2.0.5. Hence the suggestion for a documented alpha/beta process with the community.

Thanks!

svenrademakers, 2024-09-04:
Better documentation that goes more in-depth about the development processes and invites FOSS collaboration is an excellent point and is indeed currently spotty. I can't promise anything now, but I'm very open to giving someone pointers on how to update our documentation.

# development strategy

that said, the aim is to keep the development process "simple" (as long as that makes sense). We have only one "unstable" branch (main) used to tag RC's and releases on.  this means master should remain relatively stable. until now we where able to keep it stable during development)

tagging of commits looks something like

```text
commits:

* v2.1.0-unstable
* v2.1.0-1-unstable
* v2.1.0-2-unstable
*  ...
* v2.1.0-RC1
* ..
* v2.1.0
* v2.2.0-unstable
```

# trying out and reverting

since we are quite stable, its recommended to always install latest master on your board, if you want to test out new stuff. unless we explicitly made an intermediate tag. use the packages that are attached to the associated github action build. these packages you can install easily via the 'upgrade firmware tab' in the web UI

you can always easily go back and forth between all versions (bar removing any user files that conflict with a particular version). install packages are self contained and not incremental in any way

nemith, 2024-09-04:
I've been building and running pretty much the latest firmware with very little issues so far (and a lot of wonderful improvements)

rjbrown99, 2024-12-30:
Question - are you still keeping this repo current with firmware code updates?

Asking as I now have an original TPi2 board and a recently acquired 2.5.1 board. The new board arrived with daemon version 2.3.2, however I don't see that as an available tag in this repo. 2.1.0-RC2 is the most recent tag.

My goal was to update my original board so I'm running the same BMC version on both. Any suggestions? Thanks!

[image]

svenrademakers, 2025-01-03:
@rjbrown99 in the beginnings of the firmware, the fw version == bmcd version. which doesnt really hold true anymore. the bmcd is just a part of the whole fw, so can be updated differently to reflect its changes better. Fw 2.1.0-RC2 packs bmcd ver 2.3.2 for instance. hope that clears up some things.

rjbrown99, 2025-01-03:
Thank you, that's helpful.

I built the 2.1.0-RC2 firmware and installed it on my original 2.4 board. API version was 1.2 and BMC UI was 3.3.2, whereas the firmware shipped with my 2.5.1 board uses API version 1.1 but BMC UI 3.3.3. 

My question is this: should I expect any issues if I replace the firmware on the 2.5.1 board with the version I compiled (2.1.0-RC2)? That means the API will move to 1.2 but the UI moves back to 3.3.2. 

What I'm really asking is whether or not there are any specific minimum requirements for the BMC firmware for 2.5.1 boards. Thanks!

rjbrown99, 2025-01-03:
One additional note - when upgrading from firmware 2.0.5 to 2.1.0-RC2, bmcd would not start. That is because the config.yaml was missing the line:

`host: "::"`

Once I added that line the bmcd started normally. I'm guessing this was introduced somewhere between versions, but this might be worth adding to the docs somewhere.

dgdonovan, 2025-01-04:
@rjbrown99 in reality the firmware on your Turing Pi v2.5.2 board was factory installed via the image produced by the "factory" branch. If I recall correctly, the "master" branch was 7-ish commits behind the "factory" one. Some of these commits were specific to the factory test process, but some were for stability. Technically, you can install the BMC-Firmware v2.1 release candidate firmware on a v2.3/v2.4 board, BUT the eeprom will not be programmed. To see this, on the BMC of the v2.3/v2.4 board, run "tpi eeprom get". Many of the fields will have strange contents. It is possible to resolve, but isn't a process to post here. I thought I had bricked my v2.4 board. It was recoverable, but the steps are not documented. Best bet is to wait until an official release is available, but that's just my 2 cents.

svenrademakers, 2025-01-04:
@rjbrown99 technically you can down and upgrade without any limitations unless you stay above v2.0.0.
There is a caveat though:
When you touch files on the bmc that came with the stock firmware, they get copied to the user overlay partition and overlayed over the actual file. This means that from that point onwards updates to that particular file are not visible anymore. My hypositis is thats what happened in your case with the config.yaml.

I dont have a realy good migration strategy yet, other then making sure we dont make too much breaking changes :)