turing-machines/BMC-Firmware #221: EMC2301 fan controller not recognized

nmaas87, 2024-09-09:
Dear @CFSworks - thanks for getting the EMC2301 into the firmeware with #125 :).
I just added a EMC2301 to my TPi 2.4 and booted the board up, sadly there was no hwmon1 entry.
Looking at i2cdetect -y0, it seems to recognize the chip with its main address at 0x2f and its alert at 0x0c.

````
[root@turingpi 6.8.12]$ i2cdetect -y 0
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- 0c -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- UU
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: UU -- -- -- -- -- -- -- -- -- -- -- UU -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
````

It also seems to have loaded something as 0x2f is used by the kernel, but looking at lsmod, there seems to be no module loaded?

````
[root@turingpi 6.8.12]$ lsmod
Module                  Size  Used by    Tainted: G
rockusb                12288  0
````

However, the kernel still seems to have the module builtin...

````
[root@turingpi 6.8.12]$ cat modules.builtin | grep emc
kernel/drivers/hwmon/emc2305.ko
````

Any ideas on how to get it working? Currently running tp2-v2.1.0-RC2-7-g0cd2260c which is the latest firmeware built from the last commit as of now ( 0cd2260c80be4bf36b8126722f8a6a0bca13c0cf )

Thank you,
Nico

CFSworks, 2024-09-09:
Hi Nico!

It does indeed look like your EMC2301 is soldered correctly (on the I²C side, anyway!) but there might be some confusion caused by 2.1.0's bump to the latest Linux kernel. `hwmon0` is the current identify of the fan controller, and the `pwm1` file within controls it. You should also be able to control it through the WebUI and/or by writing to `/sys/class/thermal/cooling_device0/cur_state`.

A few notes that may help:
- The EMC2305 driver (which also handles EMC2301) is builtin, as you've noticed. Builtins do not appear under `lsmod`, which shows dynamic modules only.
- This latest firmware version does not enable the BMC's builtin temperature sensor. I'm still trying to figure out *why,* because the driver is there, but the (upstream) patch that was supposed to tell Linux about the existence of the thermal sensor never landed. This sensor, when enabled, would likely occupy the `hwmon0` slot and bump the fan controller to `hwmon1` as expected.
- `cooling_device0` is a more stable identifier of the fan. The `tpi cooling` command and/or the WebUI appear to be the preferred way(s) to manage it.

Regards,
Sam

nmaas87, 2024-09-12:
Thanks a lot Sam for all the information, that really worked! :). Yes I can control the fan - and finally see the slider in the webfrontend (to be honest, I did not check and was not even aware that something should be there, that was a nice surprise!). I can confirm everything works and I can also confirm the oddities you menitioned with the TPi 2 builtin temperature sensors being missing and the EMC2301 having taken the slot of hwmon0. If you have any idea or a fix to test at some point, just let me know :).

Thanks a lot,
Nico

nmaas87, 2024-11-20:
Thank you @CFSworks I added a blog post to give more details for people who want to do your little "hack" :) https://www.nico-maas.de/?p=2702