turing-machines/BMC-Firmware #224: BMC daemon fails to start - Empty SSL Certificate Files

abdavid, 2024-10-08:
Hi!

I just received my TuringPi today, and while setting everything up, I must have been too eager to remove power from the board. This caused what I identified as a state that required manual recovery.

The issue I encountered was related to the BMC’s certificate files. Specifically, I found that the `bmcd_cert.pem` file was 0 bytes in size after the system restarted. This prevented the BMC daemon and web interface from starting correctly and caused API commands such as `tpi power on` to fail.

Upon reviewing the BMC startup script, I noticed that the logic only checks for the existence of the .pem files but does not verify that the files are valid or non-empty.

Suggested Fix:

It would be ideal to improve the script by also checking that the certificate and key files are non-empty (i.e., contain valid data), using the -s flag instead of just -f. This ensures that the BMC can properly recover from situations where certificates are truncated to 0 bytes.

For example, this line:

https://github.com/turing-machines/BMC-Firmware/blob/0cd2260c80be4bf36b8126722f8a6a0bca13c0cf/tp2bmc/board/tp2bmc/factory_overlay/upper/etc/init.d/S94bmcd#L12

Could be changed to:

```bash
if [ ! -s "$CERTFILE" ] || [ ! -s "$KEYFILE" ]; then
    /etc/bmcd/generate_self_signedx509.sh
fi
```

This modification would ensure that the certificate files are checked not only for existence but also for being non-empty, helping to avoid issues like the one I encountered.

You could, of course, also do some of this validation in the bmcd project and just delete the certificates if they are not valid—whichever you find appropriate. I still don’t have a complete overview of the project in its entirety, so bear with me.

Thanks for your help, and I’m looking forward to getting everything up and running smoothly!

svenrademakers, 2024-10-15:
@abdavid thanks for reporting this! I think the appropriate thing to do here is to remove the certs if they are invalid. (that is if we know for sure they are not supplied by the user.
Let me see if i can do something about this

gjrtimmer, 2025-01-27:
@svenrademakers Do you want me to create a PR for this?

I was thinking about the following:

1. Check if CERT and KEY exists
2. If they do not exists then generate new ones
3. If only one of them exists, assume ERROR; delete, and regenerate the original
4. If both exists, then size file, if 0 bytes, assume ERROR; delete and regenerate
5. if both exists and not 0 then use `openssl` command to validate CERT and KEY, on any error, delete and regenerate

Let me know what you think about this, if this is ok, I will create a PR for this.

svenrademakers, 2025-01-27:
Sounds like a good improvement!

gjrtimmer, 2025-01-28:
I will create the PR.

gjrtimmer, 2025-01-28:
PR created; will await review, all options of the script have been tested locally by hand, no automated test created.

PR #234 (2025-02-03): fix: revert empty or faulty ssl certificates by regeneration
fixes #224

This PR updates the generate certificate script as outlined in issue #224
commit 62ca5792 (2025-02-03): feat: revert empty or faulty ssl certificates by regeneration

fixes #224
closed 2025-02-03 by svenrademakers